FreeCAD/FreeCAD #19810: Materials: Allow user to search for a material

raindropsfromsky, 2025-02-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Currently, the user cannot quickly locate the material he has in mind, such as "plastic", "gold", "steel", or "glass".

Please provide two search fields as shown in the screenshot below, to let him locate the material he wants.
[image]

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40317 (Git) Conda
Build type: Release
Branch: main
Hash: b67ab979afdba808fc03196faf9c5f8830df471f
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical/physical DPI: 96/123.928
Installed mods: 
  * Beltrami 1.1.0
  * CommandPanel
  * CurvedShapes 1.0.13
  * Curves 0.6.58
  * ExplodedAssembly
  * fasteners 0.5.37
  * frame 0.1.1
  * FrameForge 0.1.2
  * FreeCAD-Ribbon 1.8.0
  * freecad.gears 1.3.0
  * FreeGrid 2.2.0
  * OpenTheme 2024.9.1
  * parts_library
  * Quetzal 1.2.2
  * SearchBar 1.3.3
  * sheetmetal 0.7.11
  * Silk 0.1.5
  * symbols_library
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-24:
@davesrocketshop FYI

davesrocketshop, 2025-02-24:
This is planned. Please assign to me

Dave-B, 2025-07-08:
This would be a great addition.

A nice-to-have would be if it worked as find-as-you-type - filtering the tree-view to only show items that match the search, rather than the older find-on-enter search.