FreeCAD/FreeCAD #19817: FEM: Errors in exporting geometry to Elmer - boundary names replaced/lost (ElectroStatic)

mvandeg-git, 2025-02-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When exporting a geometry for simulation, part of the names get lost for boundary conditions and hence, the simulation does not run correctly.

The mesh.unv file contains the same names as in FreeCad, e.g. Face1 and Face4.
The generated case.sif file also contains "Face1" and "Face4".
However, in entities.sif and names.sif, these names are replaced by bnry3, bnry8, etc.

By editing case.sif manually, it is possible to successfully perform the simulation. The "bnry" names and original ("Face")  names can be linked by looking up the indices in the mesh.unv file and substituting the corresponding "bnry" names in the case.sif file. So all information still seems to be present, just the naming gets distorted.

Although this issue seems related to #13490, it is different. There are no non-unique names, but extra names generated for structures with existing names, that break the link between case.sif and geometry/Elmer mesh.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Dark/Fusion
Installed mods: 
  * Curves 0.6.51
  * fasteners 0.5.31
  * Render 2024.11.22
```

### Subproject(s) affected?

FEM

### Anything else?

[TwoWires.zip](https://github.com/user-attachments/files/18939212/TwoWires.zip)

The attached file contains an example geometry for which this is an issue.
The material for wires is "Copper-Generic". This does not contain a Relative Permittivity parameter by default. For a quick test, adding "Relative Permittivity = 10000" to the "Copper-Generic" Material sections in the case.sif file should give good enough results.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

FEA-eng, 2025-02-24:
@marioalexis84 FYI

marioalexis84, 2025-02-24:
I cannot reproduce.
In my case the names "Face1", "Face4" are assigned correctly in the files `entities.sif` and `mesh.names` (there is no `names.sif` file) and the simulation performs successfully (adding relative permittivity to the copper materials.)
Note that the `bnry*` names are assigned internally by `ElmerGrid` to some set of elements. `FreeCAD` has no influence on the generation of those names.

marioalexis84, 2025-02-27:
@FEA-eng  can you reproduce this?

FEA-eng, 2025-02-28:
> [@FEA-eng](https://github.com/FEA-eng) can you reproduce this?

When I run it in 1.0 on Windows (after adding relative permittivity to copper), I get some results that look normal:

[image]

[image]

In the entities.sif file, I have this:

```
!------ Skeleton for body section -----
Body 1
  Name = Solid2
End

Body 2
  Name = Solid1
End

Body 3
  Name = Solid3
End

!------ Skeleton for boundary section -----
Boundary Condition 1
  Name = Face4
End

Boundary Condition 2
  Name = Face1
End

Boundary Condition 3
  Name = bnry7
End

Boundary Condition 4
  Name = bnry8
End
```

And in the mesh.names file:

```
! ----- names for bodies -----
$ Solid2 = 3
$ Solid1 = 5
$ Solid3 = 6
! ----- names for boundaries -----
$ Face4 = 2
$ Face1 = 4
$ bnry7 = 7
$ bnry8 = 8
```

So the "bnry" names do appear but "Face1" and "Face4" are also here.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```