FreeCAD/FreeCAD #19820: [Core] Assigning properties to multiple selected objects.

kaiwas, 2025-02-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Currently, there is no way to assign properties to multiple objects at once (which already exist).
We have to create a property separately for each object.

The addFC workbench has this capability, but it would be better to have this capability at the core level (expression input is not yet available).

You can see in the video how convenient this capability would be.

https://github.com/user-attachments/assets/b71db4c5-c40a-4a27-b6c8-4460b49cbc3e

Probably, some button is required in the interface, when clicked, a dialog similar to adding a single property appears (or as when creating a VarSet).

The general principle is as follows.
1) Select the desired objects.
2) Click the button in the interface.
3) Set the property type, group, and name.
4) If necessary, enter an expression or some value.

[image]

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40317 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: b67ab979afdba808fc03196faf9c5f8830df471f
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Logical/physical DPI: 96/91.7938
Installed mods: 
  * FreeCAD_SketchArch
  * PitchedRoof
  * Road 2025.2.6
  * addFC
  * freecad.gears 1.3.0
  * sheetmetal 0.7.11
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-05-05:
Note: please also be mindful of:  
* #14633

Joao-A-Neves, 2025-05-05:
Thanks for the orientation! I will prepare a proposal with the advices given.

pieterhijma, 2025-05-08:
Hi @Joao-A-Neves, great that  you want to work on this. It would be good if we can coordinate a bit, since I'm currently working on the VarSet dialog as well.

Joao-A-Neves, 2025-05-08:
Proposal: Add "Add Property to Selected Objects" Button to the GUI.

Introduce a new button labeled “Add Property to Selected Objects”, placed next to the existing “Create a Variable Set” button in the interface.

Clicking this button would open a dialog box based on the consensus outlined in the issue #14633 (https://github.com/FreeCAD/FreeCAD/issues/14633). The dialog allows the user to define name, group, type, value and documentation. It includes three buttons: cancel, add, add and create next.

The feature will reuse the logic from the addFC workbench (https://github.com/GS90/addFC/tree/main), which allows adding properties to multiple objects by iterating through them. In freeCAD, in the GUI it would iterate the selected objects and call addDynamicProperty() on each object. Before it starts adding the new property, if one or more of the selected objects already has the property it displays an error message that says the names of the objects that already have that property, else it adds the new property to them.

kaiwas, 2025-05-08:
> logic from the addFC

One peculiarity has been revealed.
Nesting can affect the result.
AddFC algorithms work great in PARTDESIGN, since there properties are assigned to the body (upper level).

But for example in BIM or PART, if we want to assign properties to underlying (in the view tree) objects - assigning a property will assign it to the upper object, even if it was not selected.

We need to check this and not repeat the implementation one to one.