FreeCAD/FreeCAD #19826: Assembly: Joint only recomputes if forced to by me.

philjynx, 2025-02-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

[image]
The highlighted joint has its distance set by a parameter in a variable set.
Except that it doesn't. Changing the parameter has no effect on this joint unless I explicitly tell Freecad to update the joint using "Recompute object".

[image]

Here's (some of) the variable set:

[image]

Changing "Switch Distance From Centre Wall" in the variable set, has the expected immediate effect.

Steps to reproduce:
With the assembly active -
In WelderBoxParams - change the value for Switch Distance From Centre Wall and click away from that parameter.
The model of the switch will move in relation to "centre wall".
In WelderBoxParams - change the value for Switch Base Height. Click away from it. Nothing will happen.
Now, right click the joint in the tree "SwitchBaseHeight" and click recompute object. Assuming you actually changed the value in WelderBoxParams, the switch height will change.
The other options to recompute the document or the assembly have no effect on this joint.

I have tried deleting and then recreating the joint to no avail.

[18650Parts.zip](https://github.com/user-attachments/files/18945028/18650Parts.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * FreeCAD-themes 2024.7.24
  * Help 1.0.3
  * HiResMonitor
  * kicadStepUpMod 11.4.4
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-25:
@pieterhijma @PaddleStroke FYI

philjynx, 2025-03-07:
This seems to confirm that the problem is with VarSet not Assembly.
I changed the offending variable in the joint 'SwitchBaseHeight' to point to a newly added property 'swbm' of 'Welderbox' part container.

Modifying that variable has the immediate effect of correctly moving the item in the assembly.

[18650PartsRevised.zip](https://github.com/user-attachments/files/19126042/18650PartsRevised.zip)

[image]

[image]

The workaround for me seems to be to stop using VarSets.

drwho495, 2025-03-15:
This should be fixable by running purge touch any time a change to offset or similar is made. AFAIK (at least as it shows in JointObject) recomputation of joints doesn't do anything so this should be fine.

philjynx, 2025-03-15:
@drwho495 
I've asked the internet what "purge touch" means. No results served to explain your remark. 

In any case, the purpose of raising the issue is to prompt a fix at some point to freecad.

drwho495, 2025-03-15:
> @drwho495 
> I've asked the internet what "purge touch" means. No results served to explain your remark. 
> 
> In any case, the purpose of raising the issue is to prompt a fix at some point to freecad.

It means to not trigger a recompute on an object that was previously marked to recompute (or touched as its known in the code)

philjynx, 2025-03-15:
Tell freecad not to recompute the object when it is already not doing so? This will somehow be interpreted as "the number representing the location of the object has changed, so please redraw it in its new position". 

We're all doomed.

drwho495, 2025-03-15:
> Tell freecad not to recompute the object when it is already not doing so? This will somehow be interpreted as "the number representing the location of the object has changed, so please redraw it in its new position". 
> 
> We're all doomed.

Well the joint doesn't really do anything during a recompute, all changes are made after a property is changed and when you solve the joint. I'll add it to my fork to show you what I mean

drwho495, 2025-03-15:
> The very point of raising this issue is that changing the property (via the varset) does not effect a change. 
> 
> Adding "it" to your fork won't show me what you mean. I don't mean that impolitely, I'm running the version stated at the beginning of this thread.
> 
> Although I'm a competent (retired) programmer/analyst, in the context of freecad I'm an end user, not a developer.

So changing the property doesn't change the distance after a solve?

philjynx, 2025-03-15:
See original post.

drwho495, 2025-03-15:
Looking into it more shows that it is for sure a varset issue.

pieterhijma, 2025-03-17:
Thanks for reporting this. This is a very interesting one and seems to show that something is going wrong with dependencies.

```python
doc = App.ActiveDocument
def showList(list):
    return [item.Label for item in list]

doc.VarSet002.Label
showList(doc.VarSet002.InList)
```

This shows that `SwitchBaseHeight` is a dependency which is good. It also shows that `SwitchDistanceFromCentreWall` is a dependency as well and that is also good. 

When you right-click on the document `18650Parts` and checkmark "Skip recomputes", you can inspect how document objects are recomputed.

When you change `WelderBoxParams.SwitchBaseHeight` and right-click on `WelderBoxParams` to do "recomputeObject", you can see that both `SwitchDistanceFromCentreWall` and `SwitchBaseHeight` are marked to be recomputed which is what we expect.

The order does not matter at this stage. From FreeCAD's point of view, those two are independent and can be recomputed in any order, which should be true. Now, when you right-click on `SwitchDistanceFromCentreWall` to do "recomputeObject", the following interesting happens: `SwitchBaseHeight` is unmarked to be recomputed. However, this means that it will actually not be recomputed which means that it won't have the opportunity to fetch the new value from the VarSet. This also means that the assumption that FreeCAD has that the two document objects are independent does not hold. They somehow rely on each other, because otherwise recomputing one joint would not affect the other.

My guess is that somewhere in the Assembly code for this joint there is a purge of the recompute mark for the other joints. 

@PaddleStroke, can you explain why a recompute of a distance joint would prevent recomputes of other distance joints?

philjynx, 2025-06-26:
Why isn't this marked confirmed when the above discussion seems to indicate that it has been confirmed?