FreeCAD/FreeCAD #19827: [Assembly] Distance joint resolution changes after pressing Ok

mgth, 2025-02-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When creating a distance joint on a non constraint part, while the task panel is open, the proposed resolution is quite intuitive and would be nice. But as soon as I press Ok, the resolution swap to another solution. This solution is still valid, but is always worse (with more movements needed)

the assembly before adding distance joint :
[image]

the joint resolve while task is open :
[image]

the joint after pressing Ok :
[image]

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40317 (Git) Conda
Build type: Release
Branch: main
Hash: b67ab979afdba808fc03196faf9c5f8830df471f
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/79.7968
Installed mods: 
  * CurvesWB 0.6.12
  * Defeaturing 1.2.3
  * ExtremeProDark 2.6.4
  * fasteners 0.5.37
  * FreeCAD-Ribbon 1.7.3.7
  * FreeCAD-themes 2025.1.7
  * freecad-xr-workbench 0.9.4
  * freecad_streamdeck_addon 0.1.7
  * OpenDark 2024.3.13
  * OpenTheme 2024.9.1
  * parts_library
  * parts_library.bak
  * ProDarkThemePreferencePack 1.0.0
  * SearchBar 1.4.0
  * workfeature
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-25:
Is it the same in the stable 1.0 version?
Please add an example file.

mgth, 2025-02-26:
Yes it is the same in v1.0.0

here is an example, just try to join the two facing faces with a distance joint.
[BugTest.zip](https://github.com/user-attachments/files/18979545/BugTest.zip)

closed 2025-09-01 by chennes