FreeCAD/FreeCAD #19829: IfcOpenShell version corrupted when copying About info to clipboard twice

furgo16, 2025-02-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Navigate to Help > About FreeCAD
2. Click on Copy to clipboard
3. Paste the copied information somewhere and examine it => IfcOpenShell version is 0.8.1.post1 (OK)
4. Click again on Copy to clipboard
5. Paste the copied information somewhere and examine it => IfcOpenShell version is \.8.1.post1 (NOK)

Afterwards, this affects any Python code that checks for the IfcOpenShell version, e.g. the BIM workbench. After the steps above this can also be reproduced on the Python console:

```python
>>> import ifcopenshell
>>> ifcopenshell.version
'/.8.1.post1'
```

The corrupted character can be `\`, `-`, or `.` at different times. It always seems to affect the first digit (major version.

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40097 (Git)
Build type: Release
Branch: main
Hash: aef46ad56d9536ced63ce5f39d0ffe69edfca34b
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, IfcOpenShell 0.8.1.post1, OCC 7.8.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Logical/physical DPI: 96/159.34
Installed mods: 
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-02-25:
@furgo16 as a workaround why don't you change

```
    const char* cmd = "import ifcopenshell\n"
                      "version = ifcopenshell.version";
```

to:

```
    const char* cmd = "import ifcopenshell\n"
                      "version = ifcopenshell.version_core";
```

the developers will still be able to tell which version the user has installed without corrupting the `ifcopenshell.version` command in the console.

furgo16, 2025-02-25:
Thanks, that's an interesting thought. I didn't know `version_core` existed.

I'm unsure about it, though. For instance, for `version` `0.8.0`, `version_core` is `0.7.11-03935a91d`. Even when corrupted, `*.8.0` is more recognizable than `0.7.11-03935a91d`. That is, it's easier to correlate to the PyPi releases at https://pypi.org/project/ifcopenshell/#history and then identify the correct version. If I were given a `0.7.11-03935a91d` version string, I wouldn't know where to find that version.

I appreciate the fact that the workaround would have the advantage of leaving the version untouched. But as annoying this bug is, it has limited impact. The version check is used only in one place in the code, in the BIM command to manually upgrade IfcOpenShell. I'm not sure many people are using that command, particularly as it has some issues. I'm proposing to implement a workaround at https://github.com/FreeCAD/FreeCAD/issues/19821

furgo16, 2025-02-25:
[This forum post](https://forum.freecad.org/viewtopic.php?p=234853&sid=236e7543efd2797917f5135b182711c9#p234853) has an alternative approach.

Forum (snippet modified to use the current variables)

```cpp
PyObject* ifcopenshellVer = Base::Interpreter().getValue(cmd, "version");
PyObject* objectsRepresentation = PyObject_Repr(ifcopenshellVer);
const char* ifcopenshellVerAsStr = PyString_AsString(objectsRepresentation);
```

vs. the one in the code (snipped simplified for comparison):

```cpp
ifcopenshellVer = Base::Interpreter().getValue(cmd, "version");
const char* ifcopenshellVerAsStr = PyUnicode_AsUTF8(ifcopenshellVer);
```

PR #19861 (2025-02-26): Do not decrease reference counting for string
If I understand it correctly, as `ifcopenshellVerAsStr` is not a PyObject, there is no need to decrease reference counting. See also the [PyUnicode_AsUTF8 API documentation](https://docs.python.org/3/c-api/unicode.html#c.PyUnicode_AsUTF8).

Fixes #19829
closed 2025-02-26 by chennes