FreeCAD/FreeCAD #19836: [PD] Transformation tool doesn't show grabber for Clones

steveklett, 2025-02-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I've noticed when working with the new Transformation tool with Clones that there isn't a Grabber displayed.
The attached files demonstrates this simply.

**Reproduction Steps:**

1. Select Body and activate Transformation Edit > Transform
2. Note the grabber is visible
3. Select the Clone and activate Transformation Edit > Transform
4. Note no grabber visible

[no grabber for clone.FCStd.tar.gz](https://github.com/user-attachments/files/18950049/no.grabber.for.clone.FCStd.tar.gz)

### Full version info

```shell
OS: Linux Mint 22 (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.40253 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: e16c462916a050909ad107e4f1ac2e0e5990389d
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96.0135/108.888
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-25:
Does it work in 1.0?
Related: https://github.com/FreeCAD/FreeCAD/issues/19542

maxwxyz, 2025-02-25:
@kadet1090 FYI