FreeCAD/FreeCAD #19840: Deselected components still drawn as selected

steveklett, 2025-02-25:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I cannot reproduce this with a simple test file, I'm only experiencing this in my file but it is reproducible.

[image]

When selecting a list of sub components (e.g., edges) using the CTRL key modifier, then deselecting an edge it is still drawn in the 3D view as selected. The Selection list panel correctly updates, but the 3D view does not.
Here is a simple video that demonstrates this:
https://youtu.be/MYyjJMUVyhg

I tried reproducing this with a new file and a simple PAD, but it works correctly. I've attached my file that exhibits the issue.

[deselected objects still drawn selected.FCStd.tar.gz](https://github.com/user-attachments/files/18956462/deselected.objects.still.drawn.selected.FCStd.tar.gz)

### Full version info

```shell
OS: Linux Mint 22 (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.40317 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: b67ab979afdba808fc03196faf9c5f8830df471f
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96.0135/108.888
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-25:
I cannot reproduce, can you reproduce in the stable 1.0 version with your file?

steveklett, 2025-02-25:
I can reproduce with my file in stable and in build #40317