FreeCAD/FreeCAD #19854: Tab Bar for multiples files is not enough.

leoheck, 2025-02-25:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The tabbar is not enough for multiple files.
Especially when designing using Assemblies and Linked files.

When we open an Assembly, for instance, FreeCad opens all the linked files (this is a bit annoying actually) and then I have the tab bar full of tabs that are not easy to find since the only way of selecting the files quickly is to scroll to the tabs horizontally. Then I have to pass one by one this way.

I am thinking a widget that can be stacked with the Model with a list of the opened files can help here. But not as in the default view where files are mixed in the model list. This default way is confusing. But having another tab, in the model widget, where you can select easily the working file in a vertical list. And also, maybe this can work for when you have multiple linked files only... So the whole thing behaves like a project instead of just spare files. And then the tabs are used for the files that the user opened manually.

### Full version info

```shell
Version: 1.1.0dev.40317 (Git) Conda AppImage
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Zolko-123, 2025-02-27:
The "Window" menu lists all open documents. It could be made more user-friendly

leoheck, 2025-02-28:
We should discuss in having a project-like approach for Freecad. 
So the project/main file could also store variables, so they can be used everywhere in the project.
It could also be a text file, so the references could be seen/edited manually.

luzpaz, 2025-05-07:
Looks related to:  
* https://github.com/FreeCAD/FreeCAD/issues/11914

leoheck, 2025-05-07:
Related indeed, but not the same thing. 

Here, I am asking to have a way to view open files since the tabs are bad and quick to switch to files.
Long file names and many files per project make this worse since tabs start to scroll out of the view area, and also the tabs' behavior is a bit broken, when some files are opened (hidden) and not displayed either in the tabs or in the Windows menu.

Also, displaying open documents in the tree view mixed with the Objects is horrible.