FreeCAD/FreeCAD #19855: BIM_Material window too small default size

semhustej, 2025-02-25:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

**Feature request:**

Default size of BIM_Material window is too small - not all of the text is readable. See screenshot:

[image]

Proposed solution:
Make the default window width at least 450-500px, so that the text is readable

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40317 (Git) Conda
Build type: Release
Branch: main
Hash: b67ab979afdba808fc03196faf9c5f8830df471f
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell /.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/92.6073
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

yorikvanhavre, 2025-02-27:
This panel is a BIM panel, this is a BIM issue :)

Note that the panel remembers its size, so if the user resizes it, it will stay that way afterwards. But I  agree with this change. There are two places where this needs to be changed: 1) in the ui file, it's just a matter of opening it in QDesigner, resizing it and saving it, and 2) in https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/BIM/bimcommands/BimMaterial.py#L79 , change the default height and width there.

@semhustej would you want to try fixing this? 😇

semhustej, 2025-02-27:
> 
> [@semhustej](https://github.com/semhustej) would you want to try fixing this? 😇

Would this mean:

1. Cloning the FreeCAD repository
2. Chaging the value in BimMaterial.py
3. Installing Qt Designer - https://www.qt.io/download-dev (I did not find QDesigner)
4. Finding BimMaterial.ui file 
5. Opening BimMaterial.ui in the Qt Designer, changing the size according to the size changed in BimMaterial.py and saving it
6. Pushing changes to the main repository

Am I correct? I have 0 experience working with GitHub :)

yorikvanhavre, 2025-02-27:
Basically yes! With a bit more:

1. Clone the FreeCAD repo (or "sync" it if it was already cloned)
2. Create a new branch (it's easier to create a PR later)
3. Install QtDesigner (that's the one, sorry, I misspelled). It's also part of QtCreator, so you can install that one too.
4. Open BimMaterial.ui in QtDesigner/QtCreator
5. Resize the dialog to your liking (or change its height and width properties)
6. Upload the changed BimMaterial.ui to your github branch (make sure you are uploading to the correct folder so GitHub understands you are updating an existing file)
7. Make the changes in https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/BIM/bimcommands/BimMaterial.py#L79 (you can do this directly on github)
8. Github should detect that your branch can be turned into a PR, and propose you to create a PR
9. Verify that your PR is correct (only those 2 files are changed)
10. In the PR description, be sure to add "fixes #19855" so it autocloses this issue when merged
11. Congrats, you are a FreeCAD developer 😎 

Don't feel pressured to do it, I can do it too, but I thought this would be cool to try so you can help fixing these UI things when you see them ;)

semhustej, 2025-02-27:
Thanks for the list of steps to do. I can try that, although I don't know when I will get to it. But I guess there's plenty of time for this fix :).

semhustej, 2025-03-06:
Hi @yorikvanhavre ,

I decided to learn GitHub skills on a smaller project, so I created this pull request for Symblos library: https://github.com/FreeCAD/FreeCAD-symbols/pull/8

Could you please take a look at it at let me know if I can proceed in a similar way with the OP issue?

yorikvanhavre, 2025-03-06:
Yes! Good job @semhustej !

semhustej, 2025-03-08:
I am not able to find the appropriate .ui file. There is no BimMaterial.ui in the source code. And none of these seem to be the right one:

- ArchMaterial.ui
- ArchMultiMaterial.ui
- dialogMaterialChooser.ui

Is there a better way to look for the correct .ui file than search through names? 
When the size is changed in the .py file, it has the correct results, but I guess it would be nice to also change the correct .ui file for future reference.

Syres916, 2025-03-10:
@semhustej would you mind please testing https://github.com/FreeCAD/FreeCAD/compare/FreeCAD:FreeCAD:main...Syres916:FreeCAD:BIM_Materials_Ops_fix I'd rather have a second set of eyes before submitting the PR, TIA.

[image]

yorikvanhavre, 2025-03-11:
> I am not able to find the appropriate .ui file

Indeed in this case it's not an .ui file, it's entirely coded :( It's bad practice, I know 😅 but we used to do that a lot, in the old times... The best way to find out what dialog/ui is used, is first to locate where it happens: is is a command, or an object? If it's a command, they are all in the BIM/bimcommands folder. If it's an object, it's one of the files directly in the BIM folder.

In this case, it's the Material command in https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/BIM/bimcommands/BimMaterial.py

There you see the command class, and inside the Activated() function, that's what happens when you click the button.

At line 73, that's where you see the dialog is created by code, and not by importing an ui file.

The good news is, the size of the dialog is created by code, (line 79/80), so that's pretty easy to change...

semhustej, 2025-03-11:
> [@semhustej](https://github.com/semhustej) would you mind please testing [FreeCAD:FreeCAD:main...Syres916:FreeCAD:BIM_Materials_Ops_fix](https://github.com/FreeCAD/FreeCAD/compare/FreeCAD:FreeCAD:main...Syres916:FreeCAD:BIM_Materials_Ops_fix) I'd rather have a second set of eyes before submitting the PR, TIA.
> 

I tested with both light and dark theme in safe mode. Both seem fine to me.

Thank you.

semhustej, 2025-03-11:
> > I am not able to find the appropriate .ui file
> 
> Indeed in this case it's not an .ui file, it's entirely coded 

Thanks for the explanation.

PR #20138 (2025-03-12): [BIM] Material dialog - fix the minimum width of four buttons in new …
…group box

result in French:

[image]

Fixes #19855 

@semhustej has tested the PR and given positive feedback see https://github.com/FreeCAD/FreeCAD/issues/19855#issuecomment-2714407698
closed 2025-03-12 by yorikvanhavre