FreeCAD/FreeCAD #19858: BIM_IfcProperties window size

semhustej, 2025-02-25:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Feature request
The default window of `BIM_IfcProperties` tool is small so that one of the buttons has non readable text.

[image]

**Proposed solution:**
Make the default window size at least 850px wide

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40317 (Git) Conda
Build type: Release
Branch: main
Hash: b67ab979afdba808fc03196faf9c5f8830df471f
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/92.6073
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

yorikvanhavre, 2025-02-27:
I think indeed nowadays 850px is not exaggerated... 
Similar to #19855, the size must be fixed in the .ui file and in https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/BIM/bimcommands/BimIfcProperties.py#L78

semhustej, 2025-02-27:
I can try this one too, but give me some time.

semhustej, 2025-03-07:
I looked into the issue and I am quite confused with the width value.

**What I discovered:**
- BimIfcProperties.py
the width value is set to 567px

- dialogIfcProperties.ui
the width value is 616px

When I open the dialogue in safe mode in Windows, the width is ca. 750px
When I open the dialogue in safe mode in Linux-Gnome, the width is ca. 905px (with the button still not wide enough for the text to be displayed all)

**My thoughts:**
So i am confused with:
- the width in .py file is different than the default width in .ui file
- none of the values result in the default dialogue size when opened for the first time in safe mode
- different default dialogue size on different systems

It seems to me FreeCAD does not default to the width set in .py or .ui files but defaults to some minimum size which is set by minimum button sizes, not the size of the whole dialogue.

The height value seems a little more consistent.

**Solution**
@yorikvanhavre can you advise on possible fix. It seems to me the steps should be:

- change the width value in both .py and .ui files to be the same - 900px
- change the combo box sizes and button sizes in Qt Creator to be somehow fit into those 900px. But I have no idea how, because it seems to fit well to 616px width in QtCreator. It seems there is a minimum combobox width, which is the limiting factor, but I don't know how to change this minimum cobobox width.

[image]

yorikvanhavre, 2025-03-07:
Wow, that is strange. Maybe this has to do with theming?

Also, qt stores some values by itself, outside FreeCAD. For example, sometimes toolbars on/off status or dock window placements are "mysteriously saved" by Qt (on linux, it's in .config/Qt IIRC). That might be at work here.

In any case, I think I would approach this by trial and error, and find something that "works best".

Also, something not to forget, if you are compiling FreeCAD yourself: If you modify an .ui file, you need to touch the .qrc file (`touch Resources/Arch.qrc`) before compiling. Otherwise cmake does not see that you changed the file

semhustej, 2025-03-07:
I do not compile FreeCAD, I have just looked at the default values.

Yes, theming changes things - with the classic theme, the default window size is OK. But not with Light theme.

I'll try to do some trial and error :)

Syres916, 2025-03-10:
@semhustej I was helping a forum user with a macro dialog that is translated so the buttons kept getting messed up so I made them in a group box separate from everything else and now the dialog automatically adjusts to suit the language and not impacted by the stylesheet. If you want I can have a go at a fix for you?

semhustej, 2025-03-10:
> [@semhustej](https://github.com/semhustej) I was helping a forum user with a macro dialog that is translated so the buttons kept getting messed up so I made them in a group box separate from [...]

Sure, I guess that would be more robust solution than the hard-coded size. Just note that I already created a pull request which changes the default width.

Syres916, 2025-03-10:
You can try and steal/ignore (I won't be offended :smile:) as much as you want but this is what I got to work for the 3 default styles and Behave-dark from FreeCAD-themes https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:IFC_Properties_Form you may want to use a different Group Box title, I couldn't think of anything better!! I used Francais to test the button size as that tends to be a good barometer.

[image]

semhustej, 2025-03-10:
That looks good. I guess my PR is not needed now. How do I remove it so that it is not left somewhere hanging? By simply selecting 'Close pull request' here on GitHub?

[image]

semhustej, 2025-03-10:
@Syres916 will you do the same here https://github.com/FreeCAD/FreeCAD/issues/19855? I am not able to do the language-dependent adjustment that you did.

Syres916, 2025-03-10:
> That looks good. I guess my PR is not needed now. How do I remove it so that it is not left somewhere hanging? By simply selecting 'Close pull request' here on GitHub?
> 
> [image]

Yes you are OK to Close it but I'll reference it in my PR and the fact that you've tested mine.

Syres916, 2025-03-10:
> [@Syres916](https://github.com/Syres916) will you do the same here [#19855](https://github.com/FreeCAD/FreeCAD/issues/19855)? I am not able to do the language-dependent adjustment that you did.

Yep, not sure I'll get it done for this afternoon's merge though.

PR #20109 (2025-03-11): [BIM] Correct styling of DeleteIFCProperties button and add GroupBox for the two lower comboboxes and the button
Replaces https://github.com/FreeCAD/FreeCAD/pull/20070

Fixes #19858

@semhustej has tested this PR and has given positive feedback, (see screenshot in the comment previous to) https://github.com/FreeCAD/FreeCAD/issues/19858#issuecomment-2710631042 for screenshot in French

FYI @yorikvanhavre