FreeCAD/FreeCAD #19862: No Bit Tutorial in FreeCAD Help

keycal, 2025-02-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The wiki for the Bit Tutorial states that the tutorial is not to be run from the wiki.  It should be a selection from the FreeCAD "help" menu but I have no such entry.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * Dracula 0.0.5
  * Freecad-Built-in-themes-beta 1.2.2
  * FreeCAD-themes 2024.7.24
  * OpenTheme 2024.9.1
  * ProDarkThemePreferencePack 1.0.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-26:
Which bit tutorial from the wiki, please provide a link.

Roy-043, 2025-02-26:
This is about:
https://wiki.freecad.org/BIM_ingame_tutorial

I think @keycal forgot to switch to the BIM Workbench. I have tweaked a sentence in the 1st paragraph of the tutorial to clarify this.

closed 2025-02-26 by maxwxyz