FreeCAD/FreeCAD #19863: [Spreadsheet] Cursor does not change its appearance when increasing the cell size

kaiwas, 2025-02-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The cursor does not change its appearance when the mouse is not on the border of a column or row.
Sometimes the appearance changes, but at an incomprehensible moment and after that, it is impossible to change the cell size. Instead, the column or row is selected.

Video demonstration

https://github.com/user-attachments/assets/1e645e43-2866-47bb-a100-17d3e9dd40b3

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40317 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: b67ab979afdba808fc03196faf9c5f8830df471f
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Logical/physical DPI: 96/91.7938
Installed mods: 
  * FreeCAD_SketchArch
  * PitchedRoof
  * Road 2025.2.6
  * addFC
  * freecad.gears 1.3.0
  * sheetmetal 0.7.11
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

Spreadsheet

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

furgo16, 2025-02-26:
I can confirm it as well. It makes it impossible to adjust the cell width or height.

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40419 (Git) Snap 1314
Build type: Release
Branch: main
Hash: cf477695f9662d9892d5fc79e83065b4ac3d52bf
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 7.1.1, IfcOpenShell 0.8.0, OCC 7.7.1
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/95.1083
Installed mods: 
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
  * fcVM
  * ezydxf
  * Arch
  * Assembly4 0.50.17
```

kaiwas, 2025-02-26:
I was informed that an algorithm for automatically increasing the cell width when double-clicking on a delimiter was recently added.
I don't know the issue number, but maybe it's somehow related

benj5378, 2025-02-28:
I did a lot of debugging on this using GDB.

So when one hovers the column border, `setCursor` is called as expected in [qheaderview.cpp#L2275](https://github.com/openwebos/qt/blob/92fde5feca3d792dfd775348ca59127204ab4ac0/src/gui/itemviews/qheaderview.cpp#L2275), though the cursor doesn't change. And those rare times, that the cursor actually is set, it's not by `QHeaderView`.

I have no idea, where this set/unset are performed. I feel like these are one of these rabbits hole, where it's not worth for me to fix it. Already spent 2 hours on this...

xtemp09, 2025-02-28:
1. Place the mouse pointer at a header
2. Move the pointer to the cell area
3. Move the pointer to a header

In this case the cursor will change to `Qt::SplitVCursor` or `Qt::SplitHCursor`. =/

furgo16, 2025-02-28:
> I have no idea, where this set/unset are performed. I feel like these are one of these rabbits hole, where it's not worth for me to fix it. Already spent 2 hours on this...

As an alternative approach, perhaps it might be easier trying to find the first build where this bug appeared?

PR #20786 (2025-04-16): [Spreadsheet] Add usage of override cursor in SheetViewHeader
Closes #19863
commit 23e4fe2d (2025-04-16): [Spreadsheet] Add usage of override cursor in SheetViewHeader (#20786)

* [Spreadsheet] Add usage of override cursor in SheetViewHeader

Closes #19863

* Update src/Mod/Spreadsheet/Gui/SheetTableView.cpp

* [pre-commit.ci] auto fixes from pre-commit.com hooks

for more information, see https://pre-commit.ci

---------
closed 2025-04-16 by kadet1090