FreeCAD/FreeCAD #19866: [Spreadsheet] Expressions are lost when saving the file.

kaiwas, 2025-02-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The file contains a table with configuration rules. Perhaps this is the issue. If I assign two or more constraints to a skets, I lose them when saving.
When one constraint is assigned, everything works fine.

Video demonstration.

https://github.com/user-attachments/assets/29a22695-d7d5-4e06-8abd-ca5e9a5897db

The program crashed while recording the video. This happens often when I close tabs under the 3D window, but this is a topic for another issue.

The file is attached. Remove the zip extension. (You will have to add the second constraint in the same way)

[Table Configuration.FCStd.zip](https://github.com/user-attachments/files/18982381/Table.Configuration.FCStd.zip)

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40408 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: b07caa732e63064b8851ecdda184292962d95c15
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Logical/physical DPI: 96/91.7938
Installed mods: 
  * FreeCAD_SketchArch
  * PitchedRoof
  * Road 2025.2.6
  * addFC
  * freecad.gears 1.3.0
  * sheetmetal 0.7.11
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

Expressions

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-02-26:
I can reproduce the issue. I get this error when opening the file:

`106.945 <App> Document.cpp(2385): Failed to restore Table_Configuration_testing#Sketch.ExpressionEngine: Failed to parse expression.`

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40317 (Git) Conda
Build type: Release
Branch: main
Hash: b67ab979afdba808fc03196faf9c5f8830df471f
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/92.6073
```

kaiwas, 2025-11-25:
> [@maxwxyz](https://github.com/maxwxyz) blocker? (since data is lost here)

Hello!
I don't know at what stage the fix occurred, but at this point I haven't been able to reproduce the problem when creating a new file.

But! If I do the same thing in a file I attached previously, I lose ALL constraints (previously, there was only one left).

One more thing. The program is guaranteed to crash if I have rows in the configuration table. And when I close the tables tab with the last tab (see the video).
When I open the file, the constraints are saved.
In safe mode, the crash doesn't occur.

https://github.com/user-attachments/assets/cdf20f62-b050-4d84-8cf1-c5aceab99232

File
[ConfTableBug.FCStd.remove.zip](https://github.com/user-attachments/files/23742604/ConfTableBug.FCStd.remove.zip)

maxwxyz, 2026-11-07:
There seems to be a fix in AstoCAD/FreeCAD: https://github.com/AstoCAD/FreeCAD/commit/e30462ef03d2ef910631e05f29f0e90a150c880e, https://github.com/AstoCAD/FreeCAD/commit/65b90f6dd6c0da9c7ee4492a87a75e93c5a34afb. If someone wants to cherry-pick it and create a PR, these are the relevant commits.