FreeCAD/FreeCAD #19867: [Spreadsheet] Program crash when closing 3D window, spreadsheets in the document.

kaiwas, 2025-02-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

You can see the problem in the video.

The order of actions for reproduction.
1) Open the file.
2) Open the spreadsheet.
3) Close the 3D window tab.
4) Close the spreadsheet tab.

If we change the order (1,2,4,3) then there will be no crash.

https://github.com/user-attachments/assets/398505a8-813a-4ea9-aa87-18d3d5e9f8c7

The test file is attached. Remove the zip extension

[Table Configuration.FCStd.zip](https://github.com/user-attachments/files/18982692/Table.Configuration.FCStd.zip)

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40408 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: b07caa732e63064b8851ecdda184292962d95c15
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Logical/physical DPI: 96/91.7938
Installed mods: 
  * FreeCAD_SketchArch
  * PitchedRoof
  * Road 2025.2.6
  * addFC
  * freecad.gears 1.3.0
  * sheetmetal 0.7.11
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

Spreadsheet

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-02-26:
I am not able to reproduce the issue.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40317 (Git) Conda
Build type: Release
Branch: main
Hash: b67ab979afdba808fc03196faf9c5f8830df471f
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/92.6073

```

kaiwas, 2025-02-26:
> I am not able to reproduce the issue.

This doesn't reproduce even with the attached file?
Actually, I encountered this problem back in version 0.21. But I thought it was some kind of coincidence.

I also can't reproduce it if I create a simple file with a configuration table.

But I can easily reproduce the problem with any file where I have many parameters in the table.

Sometimes it doesn't reproduce on the first try. But still, it happens quite often. Two or three times is usually enough to confirm.

Here is another file of mine. (remove the zip extension)

[SuperObject_16.FCStd.zip](https://github.com/user-attachments/files/18984098/SuperObject_16.FCStd.zip)

The video shows the program crashing. But literally a minute ago, on another attempt, this did not happen.

https://github.com/user-attachments/assets/d1ce7a86-c63f-4439-8565-31117d680219

kaiwas, 2025-02-26:
I can "cure the program" once. If I do it in a new file, the question with the closing dialog appears. But if I do it again, everything breaks.

https://github.com/user-attachments/assets/d4f99484-e4d1-46f3-90d7-a1441b4a738d

semhustej, 2025-02-26:
This is my situation. I also tried in Safe mode.

Maybe the issue is Linux related? 

https://github.com/user-attachments/assets/dd447d24-0328-4209-a1fb-4ef39085ade7

kaiwas, 2025-02-26:
> Linux related?

I thought about it too, but it seems not.

The problem is not reproduced in safe mode (as with you).

I'll try to find out what could affect the result. Thanks for testing!

kaiwas, 2025-02-27:
> It seems to be an issue with the config files, could you upload them?

The configuration folder is attached.
I will try to reconfigure the program again. If it turns out that some setting gives such a negative effect, I will definitely let you know.

The problem with the configuration is that previously I had to run version 0.21 and version 1.0 at the same time. Perhaps there was some conflict.

[FreeCAD.zip](https://github.com/user-attachments/files/19002716/FreeCAD.zip)