FreeCAD/FreeCAD #19868: [Core] An unsaved file disappears from the program when another file is opened.

kaiwas, 2025-02-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

If we run the program and create a document without adding anything and without saving the document. And then open any file from the history - the first document disappears.

Expected behavior - an accessible document without a name in the tree view and in the tabs.

If we add anything to the new document - the document does not disappear.

Video demonstration.

https://github.com/user-attachments/assets/c01b65b8-6dc8-4fdf-8303-7075f6e5ac34

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40408 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: b07caa732e63064b8851ecdda184292962d95c15
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell /.7.0, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Logical/physical DPI: 96/91.7938
Installed mods: 
  * FreeCAD_SketchArch
  * PitchedRoof
  * Road 2025.2.6
  * addFC
  * freecad.gears 1.3.0
  * sheetmetal 0.7.11
  * addFC.backup1740388313.7895358 1.4.0
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-02-26:
I can confirm the issue.
```

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40317 (Git) Conda
Build type: Release
Branch: main
Hash: b67ab979afdba808fc03196faf9c5f8830df471f
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/92.6073

```

Joao-A-Neves, 2025-03-07:
Hi! I'm interested in solving this issue. Can I work on it?

semhustej, 2025-03-08:
@maxwxyz can you please advise @Joao-A-Neves ?

maxwxyz, 2025-03-08:
@Joao-A-Neves sure, you can check out https://freecad.github.io/DevelopersHandbook/gettingstarted/ on how to get started

closed 2025-04-28 by chennes

semhustej, 2025-05-17:
I did a test with the latest weekly and the fix works fine. Thank you for this contribution @Joao-A-Neves 

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41694 (Git)
Build date: 2025/05/13 12:46:54
Build type: Release
Branch: main
Hash: 5e624a4775c993a3d940f37876840c65ac0cc60c
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/126.557/1.25
Installed mods: 
  * Reporting
  * Telemetry 1.0.1

```

PR #20554 (2025-04-28): Core: Fixed a bug where an empty document disappears
Created a flag named autoCreated to distinguish an autoCreated document created in the startup from a manually document. Implemented a setter and a getter for this new flag. Added a codition that verifies if a document is autoCreated when opening another document to close it in the correct case. Implemented unit tests for theses cases. Fixes #19868.
commit 0d38b2f7 (2025-04-28): Core: Fixed a bug where an empty document disappears (#20554)

* Core: Fixed a bug where an empty document disappears

Created a flag named autoCreated to distinguish an autoCreated
document created in the startup from a manually document.
Implemented a setter and a getter for this new flag.
Added a codition that verifies if a document is autoCreated
when opening another document to close it in the correct case.
Implemented unit tests for theses cases. Fixes #19868.

* Tests: Fix failing auto-created document tests

* Tests: moved created tests to the existing Document test framework.

------- [... cut]