FreeCAD/FreeCAD #19871: Display error in TechDraw

entkon, 2025-02-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Die Darstellung ist fehlerhaft, wenn ich unter Daten, Coarse View auf true umstelle, wird das richtig dargestellt.
Sie können diese Datei unter diesem Link herunterladen.
https://entkon.de/freecad22-part-design-einfuehrung-in-klassische-konstruktion/

The display is faulty, but when I set Coarse View to true under Data, it is displayed correctly.
You can download this file from this link.
https://entkon.de/freecad22-part-design-einfuehrung-in-klassische-konstruktion/

[image]
[image]

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40176 (Git) Conda
Build type: Release
Branch: main
Hash: 2745f436026b6f1fb84dbaaf89ea22b5fb4c2295
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/81.5973
Installed mods: 
  * Curves 0.6.58
  * fasteners 0.5.37
  * sheetmetal 0.7.11
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-26:
I can confirm.
@WandererFan shouldn't coarse view true be "worse" than the precise option?

WandererFan, 2025-02-27:
At very high magnification, the coarse view would be worse.

There are 3 OCC projectors: the regular(exact maths, orthogonal) one, the coarse view (based on the shape's mesh representation) and the perspective  (also exact maths) projector.

The coarse view is fast, and as far as appearance is concerned, it should look the same as the regular view (except for all the vertices) at reasonable scales.  

It is not unheard of for the regular projector to have rendering issues where the other two do not.  The coarse view is pretty reliable, but is not very helpful if you want to add dimensions.   

The perspective projector with a long focus length also makes a nice picture.