FreeCAD/FreeCAD #19872: AddonMannager: Issue when updateing Python Modules

leoheck, 2025-02-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I came here since after installing `DynamicData WB` I saw this message here:
```
11:00:07  DynamicData updater: The requests package was not found, cannot check for updates.  Install requests package or disable auotomatic update checking in DynamicData settings to prevent this warning message.
```

And then this is the first time I am seeing this window working doing something.

Many items can be updated.
But something is happening under the hood.
The window content keeps changing/dancing.

Here it is changing the content...
https://github.com/user-attachments/assets/e6045ad8-16b2-46b2-a126-f78047e10729

Here the same, but the text inside some cells change to `Updating...`
https://github.com/user-attachments/assets/43ea2cc5-715d-4fff-9b60-131fef8b1366

In the console (out of Freecad, in the command line) I am seeing this:
```
pip timed out
pip timed out
pip timed out
pip timed out
pip timed out
pip timed out
pip timed out
pip timed out
pip timed out
pip timed out
pip timed out
pip timed out
pip timed out
pip timed out
pip timed out
pip timed out
pip timed out
pip timed out
pip timed out
pip timed out
pip timed out
```

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40317 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: b67ab979afdba808fc03196faf9c5f8830df471f
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical/physical DPI: 96/145.143
Installed mods: 
  * Assembly4 0.50.18-dev
  * sheetmetal 0.7.11
  * DynamicData 2.71.0
  * Curves 0.6.58
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
  * GDT
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.1.2
  * lattice2 1.0.0
  * kicadStepUpMod 11.5.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-02-26:
There are at least two issues in this, so let's keep this simple, the Python package `requests` is used for Http(s) lookups in this case to Github, as WebEngine was removed from Conda builds then this is a fairly educated guess that `requests` was included in it. @mwganson can this lookup use existing AddonManager code from a variety of FreeCAD versions or do you need to add `requests` as a dependency to your `package.xml`?

With regard to a workbench that has a stated dependency such as Sheetmetal which uses `networkx` I was corrected presented with the question do you wish to install it and on clicking _Yes_ it hangs at _Installing Dependencies_ messagebox using today's AppImage.

```
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.40408 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: b07caa732e63064b8851ecdda184292962d95c15
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Logical/physical DPI: 96/93.1131
Installed mods: 
  * FreeCAD-themes 2025.1.7
  * DynamicData 2.71.0
  * sheetmetal 0.7.11
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.29.8
  * fasteners 0.5.37
  * freecad.gears 1.3.0
  * Curves 0.6.58

```

mwganson, 2025-02-26:
I have updated DynamicData to version 2.72 to hopefully address this issue.  I don't think it's a matter of requests package missing, but rather for some reason is not yet available to be imported during the workbench initialization.  The workaround is to try to import it again if the previous import was unsuccessful.

Syres916, 2025-02-27:
Thanks @mwganson that's quite bizarre but works. @leoheck also the Python dependencies are installing fine today (tested using Sheetmetal with `networkx`) using:

```
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.40434 (Git)
Build type: Release
Branch: main
Hash: 0ca76cc46467fa4a92fa8a039cde9428c6573777
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, IfcOpenShell 0.8.1.post1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Logical/physical DPI: 96/93.1131
Installed mods: 
  * FreeCAD-themes 2025.1.7
  * DynamicData 2.72.0
  * sheetmetal 0.7.20
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.29.8
  * fasteners 0.5.37
  * freecad.gears 1.3.0
  * Curves 0.6.58
```

maxwxyz, 2025-02-28:
so the issue is fixed?

leoheck, 2025-02-28:
I don´t think so @maxwxyz 

```
11:44:18  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
11:44:18  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
11:44:18  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
11:44:18  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
11:44:19  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
11:44:19  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
11:44:19  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
11:44:19  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
11:44:19  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
11:44:19  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
11:44:19  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
11:44:20  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
11:44:20  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
11:44:20  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
11:44:20  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
11:44:20  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
11:44:21  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
11:44:21  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
11:44:21  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
11:44:21  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
11:44:21  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
11:44:21  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
11:44:21  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
11:44:21  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
11:44:21  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
11:44:21  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
11:44:21  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
```

furgo16, 2025-02-28:
> 11:44:18  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 

I noticed these a while ago. I seem to remember they were only warnings for assets that could not be fetched for particular add-ons. These should probably be either hidden or replaced by a message stating which asset could not be fetched. This might not be an issue with the Add-On Manager itself, but from the add-on failing to provide those assets. I'm not entirely sure about this, though.

If you enable recording Log messages under Preferences, these messages appear after downloading the `package.xml` file from an add-on. Here's an extract from starting Add-On Manager right now:

```
16:21:11  Downloaded package.xml for yaml-workspace
16:21:11  Downloaded package.xml for addFC
16:21:12  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
16:21:12  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
16:21:12  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
16:21:12  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
16:21:12  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
16:21:12  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
16:21:12  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
16:21:12  Downloaded package.xml for Cables
16:21:12  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
16:21:12  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
16:21:12  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
16:21:12  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
16:21:12  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
16:21:12  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
16:21:12  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
16:21:12  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
16:21:12  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
16:21:12  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
16:21:12  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
16:21:12  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
16:21:12  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
16:21:13  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
16:21:13  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
16:21:13  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
16:21:13  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
16:21:13  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
16:21:13  Downloaded package.xml for Ondsel-Lens
16:21:14  Addon Manager: Skipping update check because AutoCheck user preference is False

```

Syres916, 2025-02-28:
This is the first 7:

```
ContentNotFoundError - https://github.com/shaise/FreeCAD_FastenersWB             PR submitted
ContentNotFoundError - https://github.com/oddtopus/flamingo
ContentNotFoundError - https://github.com/Pegoku/FreeCAD-WakaTime                PR submitted
ContentNotFoundError - https://github.com/alekssadowski95/FreeCAD-Beginner-Assistant
ContentNotFoundError - https://github.com/FreeCAD/FreeCAD-documentation
ContentNotFoundError - https://github.com/Reqrefusion/FreeCAD-Documentation-html
ContentNotFoundError - https://github.com/APEbbers/FreeCAD-Ribbon                PR Submitted
```

leoheck, 2025-02-28:
The easiest is to make a symlink with a folder called `icon` to the right one.

leoheck, 2025-02-28:
But the error could be, image xxx is missing, then people seeing these would report for their loved add-ons as you are doing.