FreeCAD/FreeCAD #19876: Creating a LCS Crashes Freecad!

leoheck, 2025-02-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Oh god. This is a regression.

```
Case/v1.3/Design took 56.9s …
➜ freecad Parts/Body.FCStd
Using: /home/lheck/AppImages/FreeCAD_weekly-builds-40408-conda-Linux-x86_64-py311.AppImage

FreeCAD 1.1.0, Libs: 1.1.0devR40408 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
Traceback (most recent call last):
  File "/tmp/.mount_FreeCAdLnpNm/usr/Mod/Part/AttachmentEditor/TaskAttachmentEditor.py", line 362, in addSelection
    self.form.message.setText(translate('AttachmentEditor',"Ignored. Can't attach object to itself!",None))
    ^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'message'
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45330) [0x76d689445330]
#1  /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libQt5Widgets.so.5(+0x2a72cb) [0x76d68aaa72cb]
#2  /tmp/.mount_FreeCAdLnpNm/usr/lib/python3.11/site-packages/PySide2/QtWidgets.cpython-311-x86_64-linux-gnu.so(+0x4828bc) [0x76d6438828bc]
#3  /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libpython3.11.so.1.0(+0x205301) [0x76d68be05301]
#4  /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libpython3.11.so.1.0(PyObject_Vectorcall+0x33) [0x76d68be00283]
#5  /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6f2) [0x76d68bdf1532]
#6  /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libpython3.11.so.1.0(+0x235664) [0x76d68be35664]
#7  0x76d68cd99847 in Gui::SelectionObserverPythonHandler::addSelection(Gui::SelectionChanges const&) from /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libFreeCADGui.so+0x697
#8  0x76d68cd8191c in Gui::SelectionObserver::_onSelectionChanged(Gui::SelectionChanges const&) from /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libFreeCADGui.so+0x2c
#9  /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libFreeCADGui.so(+0x805e85) [0x76d68ca05e85]
#10  0x76d68cd836df in Gui::SelectionSingleton::slotSelectionChanged(Gui::SelectionChanges const&) from /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libFreeCADGui.so+0x27f
#11  /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libFreeCADGui.so(+0x805e85) [0x76d68ca05e85]
#12  0x76d68cd83a0e in Gui::SelectionSingleton::notify(Gui::SelectionChanges&&) from /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libFreeCADGui.so+0x28e
#13  0x76d68cd85b5b in Gui::SelectionSingleton::addSelection(char const*, char const*, char const*, float, float, float, std::vector<Gui::SelectionSingleton::SelObj, std::allocator<Gui::SelectionSingleton::SelObj> > const*, bool) from /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libFreeCADGui.so+0x90b
#14  0x76d68c9fc868 in Gui::DocumentItem::updateItemSelection(Gui::DocumentObjectItem*) from /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libFreeCADGui.so+0x6c8
#15  0x76d68c9fcaec in Gui::DocumentItem::clearSelection(Gui::DocumentObjectItem*) from /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libFreeCADGui.so+0x11c
#16  0x76d68ca22484 in Gui::TreeWidget::onItemSelectionChanged() from /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libFreeCADGui.so+0x134
#17  /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libQt5Core.so.5(+0x2eb6f2) [0x76d689aeb6f2]
#18  /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libQt5Widgets.so.5(+0x43cc76) [0x76d68ac3cc76]
#19  /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libQt5Core.so.5(+0x2eb964) [0x76d689aeb964]
#20  0x76d689a64ce7 in QItemSelectionModel::selectionChanged(QItemSelection const&, QItemSelection const&) from /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libQt5Core.so.5+0x27
#21  0x76d689a6d08e in QItemSelectionModel::select(QItemSelection const&, QFlags<QItemSelectionModel::SelectionFlag>) from /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libQt5Core.so.5+0x1ee
#22  0x76d68ac318e8 in QTreeViewPrivate::select(QModelIndex const&, QModelIndex const&, QFlags<QItemSelectionModel::SelectionFlag>) from /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libQt5Widgets.so.5+0x688
#23  0x76d68ac321b7 in QTreeView::setSelection(QRect const&, QFlags<QItemSelectionModel::SelectionFlag>) from /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libQt5Widgets.so.5+0x167
#24  0x76d68abc3423 in QAbstractItemView::mousePressEvent(QMouseEvent*) from /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libQt5Widgets.so.5+0x2e3
#25  0x76d68a9aec08 in QWidget::event(QEvent*) from /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libQt5Widgets.so.5+0x178
#26  0x76d68aa56c6e in QFrame::event(QEvent*) from /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libQt5Widgets.so.5+0x1e
#27  0x76d689ab68bb in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libQt5Core.so.5+0x7b
#28  0x76d68a97039e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libQt5Widgets.so.5+0x6e
#29  0x76d68a978262 in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libQt5Widgets.so.5+0x13a2
#30  0x76d68c7786a9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libFreeCADGui.so+0xc9
#31  0x76d689ab6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libQt5Core.so.5+0x108
#32  0x76d68a9763eb in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libQt5Widgets.so.5+0x1ab
#33  /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libQt5Widgets.so.5(+0x1c6e6f) [0x76d68a9c6e6f]
#34  /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libQt5Widgets.so.5(+0x1ca0c9) [0x76d68a9ca0c9]
#35  0x76d68a9703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#36  0x76d68c7786a9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libFreeCADGui.so+0xc9
#37  0x76d689ab6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libQt5Core.so.5+0x108
#38  0x76d68a1353ec in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libQt5Gui.so.5+0x84c
#39  0x76d68a10b0bc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libQt5Gui.so.5+0xac
#40  /tmp/.mount_FreeCAdLnpNm/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x64bea) [0x76d6846a0bea]
#41  /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x76d688110ad7]
#42  /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x76d688113ac7]
#43  /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x76d6881140ce]
#44  0x76d689b0aa36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libQt5Core.so.5+0x66
#45  0x76d689ab570b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libQt5Core.so.5+0x11b
#46  0x76d689abd212 in QCoreApplication::exec() from /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libQt5Core.so.5+0x82
#47  0x76d68c6d77a5 in Gui::Application::runApplication() from /tmp/.mount_FreeCAdLnpNm/usr/bin/../lib/libFreeCADGui.so+0x8e5
#48  /tmp/.mount_FreeCAdLnpNm/usr/bin/freecad(+0x68e0) [0x648029dbc8e0]
#49  /lib/x86_64-linux-gnu/libc.so.6(+0x2a1ca) [0x76d68942a1ca]
#50  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x76d68942a28b]
#51  /tmp/.mount_FreeCAdLnpNm/usr/bin/freecad(+0x6c61) [0x648029dbcc61]
```

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40408 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: b07caa732e63064b8851ecdda184292962d95c15
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical/physical DPI: 96/145.143
Installed mods: 
  * Assembly4 0.50.18-dev
  * sheetmetal 0.7.11
  * DynamicData 2.71.0
  * Curves 0.6.58
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
  * GDT
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.1.2
  * lattice2 1.0.0
  * kicadStepUpMod 11.5.2
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

leoheck, 2025-02-26:
https://github.com/Zolko-123/FreeCAD_Assembly4/issues/539

furgo16, 2025-02-26:
Could you please add the steps to reproduce?

If we're trying to do the same thing, it works here:

[image]

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40421 (Git) Snap 1315
Build type: Release
Branch: main
Hash: f9bb6508ffb420502d7b91592d87ea4f95c399d4
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 7.1.1, IfcOpenShell 0.8.0, OCC 7.7.1
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/159.34
Installed mods: 
  * Launcher
```

leoheck, 2025-02-26:
Sure, sorry.

Steps to reproduce.

1. Create a Body, or select an existing one.
2. Using the Asm4, `Create Datum Object`
3. click Ok to use the suggested name 
4. click Ok, since we can do the setup later...
5. (optional) You can refresh Frecad `Ctrl+R`
6. Click the new LCS, and it will show the Report messages.

To make it crash it is a bit more random after step 6 above.
I could make it crash 3 times while writing this. 
However, I could not understand what is necessary to trigger the crash to give you the right way to reproduce.
Maybe step 5 is required for Frecad to crash.

maxwxyz, 2025-02-26:
ASM4 issue

leoheck, 2025-02-26:
> ASM4 issue

Maybe, is it? It started today after upgrading to the latest version of Freecad

Ah, I think I got it.

New Steps to reproduce.

1. Create a Body, or select an existing one.
2. With the Body selected, using the Asm4, `Create Datum Object`
3. Ok to use the suggested name
4. click Ok, to close the setup since it is not necessary.
5. Click on the new LCS
6. and then click on another object, for instance, Origin since my Body is empty.

leoheck, 2025-02-26:
@maxwxyz it is not Asm4, it is Freecad Issue.

leoheck, 2025-02-26:
Well, I could not reproduce doing this:

Creating a `PartDesign_Body`, and then creating `Part_CoordinateSystem` inside of it without using Asm4.
It was not possible to reproduce the messages or the crash.

The LCS is being created like the Origin, with things inside of it.
Which is different than the Asm4 LCS.

So if Asm4 LCS should be equal to the `Part_CoordinateSystem` then it is not equal anymore.
If this is true, it is a problem for Asm4, indeed.

Can someone confirm this?

leoheck, 2025-02-26:
Here is the differences, Freecad LCS has these things inside.
Asm4 is inside the Body, and it is a single object not a container.

[image]

furgo16, 2025-02-26:
I still cannot reproduce the issue with the steps outlined at https://github.com/FreeCAD/FreeCAD/issues/19876#issuecomment-2685971774 using Asm4, though.

The LCS is created correctly, and selecting it does not return any errors on the Report View:

[image]

leoheck, 2025-02-26:
@furgo16 , your version is a bit different mine is `v0.50.18-dev`

furgo16, 2025-02-26:
That's the version I just installed with the Add-On Manager. Where is `v0.50.18-dev` available at?

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40421 (Git) Snap 1315
Build type: Release
Branch: main
Hash: f9bb6508ffb420502d7b91592d87ea4f95c399d4
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 7.1.1, IfcOpenShell 0.8.0, OCC 7.7.1
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/159.34
Installed mods: 
  * Assembly4 0.50.17
  * Launcher
```

leoheck, 2025-02-26:
> This is my guess, I might be wrong though. In any case, if the above is correct, we should probably add a deprecation notice on https://wiki.freecad.org/PartDesign_CoordinateSystem.

Yes, this is a good decision if this is true.

leoheck, 2025-02-26:
This version is the `development` branch. 
Can you switch branches using AddonManager? By default, it uses the `main` branch I guess.

If not possible, you can do this manually by going to the Mod/Assembly4 folder and changing the branch to `development`

furgo16, 2025-02-26:
Still no crash:

[image]

> Can you switch branches using AddonManager? By default, it uses the main branch I guess.

I think you can, but I've never done it.

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40421 (Git) Snap 1315
Build type: Release
Branch: main
Hash: f9bb6508ffb420502d7b91592d87ea4f95c399d4
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 7.1.1, IfcOpenShell 0.8.0, OCC 7.7.1
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/159.34
Installed mods: 
  * Assembly4 0.50.18-dev
  * Launcher
```

leoheck, 2025-02-26:
@furgo16 I think you wont be able to reproduce this. 

I added a change to the Asm4 to use the equation on Asm4 placement.
This was working until this latest AppImage but I think something has changed that triggered this problem.

Here is a preview of this thing.

[image]

leoheck, 2025-02-26:
Or maybe it is not related. I am a bit confused. I will have to check this better later today.

furgo16, 2025-02-26:
Do I understand it correctly that you modified the Asm4 sources? If so, it would have made sense to mention that in the original description before other people start testing it.

Happy to help if I can, but in the future, please start with adding the steps to reproduce in the description, and additional information where relevant. Otherwise it's not the best use of each other's time.

leoheck, 2025-02-26:
It is still crashing with this version. I cannot even work anymore.
@furgo16 can you show me the last commit you have?

```
commit 8e21e841bb64c7aa98acf7b0a0edc0a990670ee0 (HEAD -> development, upstream/development)
Author: Zolko-123 <zoltan.hubert@zzaero.com>
Date:   Sun Feb 16 22:29:01 2025 +0100

    fixed some minor bugs in the Origin planes color code in newPartCmd when creating a Body
```

leoheck, 2025-02-26:
Steps:

1. Select the Body
2. Create a new Asm4 LCS (it can be empty but I am selecting some edges), close the window
3. Select the Body again
4. Create a new Ams4 LCS
5. CRASH

leoheck, 2025-02-26:
WORKAROUND: Once I create an LCS, I save Freecad, then it crashes After I launch it and I can work on the newly created LCS. I needed 2, so I had to crash it twice to be able to use them.

leoheck, 2025-02-26:
@furgo16 by any chance, do you have the AppImage before this one? If yes, Can you share it with somehow?

furgo16, 2025-02-26:
No luck in having a crash. I tested different combinations of creating the Body with PartDesign or Asm4, and then adding an LCS via the FreeCAD command or via the Asm4 command. I can even create LCSs outside of the body and then drag them inside the Body container without issues. It might be that we have a different combination of dependency versions installed.

[image]

> by any chance, do you have the AppImage before this one? If yes, Can you share it with somehow?

Sorry, I don't use AppImages. I use the daily build snap (see the about info in the previous comments).

Syres916, 2025-02-27:
Root cause: https://github.com/FreeCAD/FreeCAD/commit/b1a24deac64af09c78da1a518b96c3f9475f3cc7
Fixed in: https://github.com/FreeCAD/FreeCAD/commit/742627868ea59f701f10ff6b58457be329ca049f (Git revision 40409)

leoheck, 2025-02-27:
Ah, wait, you are not in the same build then.
Mine: `Version: 1.1.0, Libs: 1.1.0devR40408 (Git)`
Yours: `Version: 1.1.0dev.40421 (Git) Snap 1315`
This is good since it seems you are in the future...

Ah, yeah @Syres916 ! Thanks.

Syres916, 2025-02-27:
@leoheck on a side note you might want to change line 607 of `configurationEngine.py` from:

`    badChars = '`~!@#$%^&*()-+=|\;:\'".,'`

to:

`    badChars = r'`~!@#$%^&*()-+=|\;:\'".,'`

if you want to keep Python 3.12+ users happy whilst maintaining backwards compatibility to Python 3.8.

leoheck, 2025-02-27:
This one @Syres916 ```$(HOME}/.local/share/FreeCAD/Mod/Assembly4/configurationEngine.py```?

leoheck, 2025-02-27:
I had to fix it with this one: It looks like there is a missing scape on your code above.

```badChars = r'~!@#$%^&*()-+=|;:\'".,'```

Syres916, 2025-02-27:
> I had to fix it with this one: It looks like there is a missing scape on your code above.
> 
> `badChars = r'~!@#$%^&*()-+=|;:\'".,'`

Yep, I didn't check the code tags before I pressed Comment but you'll be all sorted when the next AppImage is released.

leoheck, 2025-02-27:
I have no time to wait, I have to build it myself now.
I am just trying to find the right commit to start with.

leoheck, 2025-02-27:
@Syres916 do you know which branch is the current 1.1-dev?

Syres916, 2025-02-27:
main

Zolko-123, 2025-02-27:
Hello, I don't completely understand : what is the situation with this right now ?

> They're two different LCS objects:
>     https://wiki.freecad.org/PartDesign_CoordinateSystem
>     https://wiki.freecad.org/Part_CoordinateSystem
> I think the first one **is deprecated** and was hidden from the UI in 1.1.0dev, but it still exists for compatibility reasons.
> In any case, if the above is correct, we should probably **add a deprecation notice** on https://wiki.freecad.org/PartDesign_CoordinateSystem.

when are you going to stop destroying FreeCAD ?

Zolko-123, 2025-02-27:
@Syres916 :

> on a side note you might want to change line 607 of `configurationEngine.py` from:
> 
> ` badChars = '`~!@#$%^&*()-+=|;:'".,'`
>
> to:
> 
> ` badChars = r'`~!@#$%^&*()-+=|;:'".,'`
> 
> if you want to keep Python 3.12+ users happy whilst maintaining backwards compatibility to Python 3.8.

if this is a general fix for Asm4, would you be so kind and make a PR for that to the Asm4 repo ?

closed 2025-02-27 by maxwxyz

Zolko-123, 2025-02-27:
> They're two different LCS objects:
> 
>     2. https://wiki.freecad.org/Part_CoordinateSystem

that is not a coordinate system, that's something else. A coordinate system is a mathematical object with a precise definition, and it doesn't contains other objects. A coordinate system allows to define the position of points in space, that's all, and that's what the PartDesign::CoordinateSystem does. It's defined by a transformation (`Placement` in FreeCAD terminology) relative to other coordinate systems, or by a point and 3 vectors (which define the same transformation of it's an orthonormal coordinate system). A mathematical coordinate system does not contain planes and axes. 

If you don't know the mathematics behind objects you should keep your hands off them. @wwmayer : are you aware of this code ?

And why does it give a name (LCS) that is already used by some existing feature working since years ? Users will search for LCS and they will find conflicting information. What's the point in re-inventing the wheel ?

leoheck, 2025-02-27:
@maxwxyz you are wrong again closing the issue because "it is not planned". 

It should be open until I can test it (which I couldn't since the latest main doesn't give me a working software).

Or it should be closed because a *Regression was fixed*.
> Root cause: [b1a24de](https://github.com/FreeCAD/FreeCAD/commit/b1a24deac64af09c78da1a518b96c3f9475f3cc7) Fixed in: [7426278](https://github.com/FreeCAD/FreeCAD/commit/742627868ea59f701f10ff6b58457be329ca049f) (Git revision 40409)

kaiwas, 2025-02-27:
> you are wrong again closing the issue because "it is not planned".

You are not alone. There were other issues that in my opinion were closed in a hurry. Maybe there are some plans to close the issues. For example, close 30 issues per month )))

maxwxyz, 2025-02-27:
You can report back and reopen if the issue is still present.

leoheck, 2025-02-27:
Ok, but I cannot reopen it.

[image]

leoheck, 2025-03-01:
Ok,  I can confirm, using the newly available AppImage `40444`, the issue seems fixed. Thanks @Syres916.