FreeCAD/FreeCAD #19877: Wireframe view seems to not be working anymore

leoheck, 2025-02-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Shortcut `v, 3` or selecting the `wireframe view does not change the view`. The other options continue working though.

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40408 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: b07caa732e63064b8851ecdda184292962d95c15
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical/physical DPI: 96/145.143
Installed mods: 
  * Assembly4 0.50.18-dev
  * sheetmetal 0.7.11
  * DynamicData 2.71.0
  * Curves 0.6.58
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
  * GDT
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.1.2
  * lattice2 1.0.0
  * kicadStepUpMod 11.5.2
```

### Subproject(s) affected?

None

### Anything else?

OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40408 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: b07caa732e63064b8851ecdda184292962d95c15
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical/physical DPI: 96/145.143
Installed mods: 
  * Assembly4 0.50.18-dev
  * sheetmetal 0.7.11
  * DynamicData 2.71.0
  * Curves 0.6.58
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
  * GDT
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.1.2
  * lattice2 1.0.0
  * kicadStepUpMod 11.5.2

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

furgo16, 2025-02-26:
Cannot reproduce, `v, 3` works fine here:

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40421 (Git) Snap 1315
Build type: Release
Branch: main
Hash: f9bb6508ffb420502d7b91592d87ea4f95c399d4
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 7.1.1, IfcOpenShell /.8.0, OCC 7.7.1
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/159.34
Installed mods: 
  * Launcher
```

Is there anything wrong with build `Version: 1.1.0dev.40408`? I seem to see lots of issues reported with it.

Syres916, 2025-02-26:
Nothing wrong with 40408 with regard to the `v,3 `and the other shortcuts I tried, all of them were Part Wb based models as well as the `PartDesignExample.FCStd` included model.

```
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.40408 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: b07caa732e63064b8851ecdda184292962d95c15
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Logical/physical DPI: 96/93.1131
Installed mods: 
  * FreeCAD-themes 2025.1.7
  * DynamicData 2.71.0
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.29.8
  * fasteners 0.5.37
  * freecad.gears 1.3.0
  * Curves 0.6.58

```

maxwxyz, 2025-02-28:
Cannot reproduce on Windows too.

leoheck, 2025-02-28:
https://github.com/user-attachments/assets/6d586856-96fc-439f-95d4-09db4dd8a5b3

leoheck, 2025-03-01:
Tested with another project 
- it is working on 1.0
- it is also working on the latest AppImage `1.1-dev 40444` 

But the project above still does not work on any.
The file that does not have this working does not have any Bodies.
Just an Asm4 Assembly.

Another one FCSTd, with an Asm4 Assembly that has a board imported from Kicad and an Antenna, has this same problem on a single object (the antenna), but not in the imported board.

The antenna is imported from another file.
While the Board (mainboard) is in this same file.
[image]

The other views work, this one is the `Hidden Line`
[image]

The antenna itself works. 
[image]

I think it has to do with Linked Objects from Other files.

leoheck, 2025-03-01:
I am confirming it.
This is something related to Links. In this case, I am using Links with Asm4, but it should not matter.

[image]

Here is the sample file with this example.
The asm file has the link to the part_a
[test_wireframe.zip](https://github.com/user-attachments/files/19035162/test_wireframe.zip)

leoheck, 2025-03-22:
This same thing affects the visibility and the possibility of changing the color of linked objects from external files.

This bug persists on `1.1.0-40655`

```
OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40655 (Git) Conda AppImage
Build date: 2025/03/10 16:51:09
Build type: Release
Branch: main
Hash: f311bfc839a4c00eff9bfb294c4ec9f670410296
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/145.143/1
Installed mods: 
  * fasteners.backup1742217927.3036284 0.5.37 (Disabled)
  * Assembly4 0.50.18-dev
  * sheetmetal 0.7.21
  * DynamicData 2.73.0
  * Curves 0.6.59
  * InventorLoader 1.5.1
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.1.2
  * lattice2 1.0.0
  * kicadStepUpMod 11.5.5
```

leoheck, 2025-03-22:
The Material settings, show similar to `disabled`. I cannot control its colors.
This is a critical bug since it breaks the work with external files and Links.

[image]

leoheck, 2025-03-27:
I can now toggle the Transparency and also change the Colors/Materials in version `1.1.0dev.40971 (Git) Conda)`.

The issue on `Wireframe mode` persists.

leoheck, 2025-04-09:
This issue persists on `FreeCAD 1.1.0, Libs: 1.1.0devR41138`

I created a simpler sample project to show the problem. 
The main file has a Body and also Links the External_Body from another file.

[wireframe_issue.zip](https://github.com/user-attachments/files/19674005/wireframe_issue.zip)

The External_Body does not change the mode to Wireframe, as it is shown below.
[image]

Now, if I go to the External file, I can set the External_Body as a wireframe in the main file.

[image]

leoheck, 2025-04-09:
Oh, this looks the same issue. 
https://github.com/FreeCAD/FreeCAD/issues/16830

closed 2025-04-10 by maxwxyz

leoheck, 2025-04-10:
@maxwxyz the other issue doesn't work on 1.0. This issue here works with it and also with a recent weekly build as I reported above.