FreeCAD/FreeCAD #19879: Reset light preferences don't reset headlight and default is not compatible with existing config files

maxwxyz, 2025-02-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

https://github.com/FreeCAD/FreeCAD/pull/19397 introduced three-point lights and a default configuration.
When the FreeCAD 1.1dev is started, it uses the same headlight direction and backlight setting of any previously installed FreeCAD version (config file).

So in my case, or for a default 1.0, headlight at (0,0,-1) and no backlight, so all issues the new light system is supposed to fix is still there (so planar faces still appear white):

[image]

[image]

When resetting the preferences, the intensity is corrected but the headlight direction is not changed (still -90°):

[image]

When I adjust main light to have the correct default and then reset preferences, they do not change:

[image]

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40421 (Git) Conda
Build type: Release
Branch: main
Hash: f9bb6508ffb420502d7b91592d87ea4f95c399d4
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/108.888
Installed mods: 
  * Cables-master 0.1.1
  * CfdOF 1.29.8
  * CurvedShapes 1.0.13
  * Curves 0.6.58
  * dodo 1.0.1
  * fasteners 0.5.37
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.1.1
  * parts_library
  * Rocket 4.1.1
  * sheetmetal 0.7.11
  * Silk 0.1.5
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

FEA-eng, 2025-02-26:
There's one additional small issue with the new light - the tooltip in the 3D view shows that it's possible to drag the handle to adjust the direction of the light while it's no longer the case. And the 3D view is now pretty much a 2D view. So unless handles for each light are added or the sphere is made movable/rotatable, the 3D view should be updated.

FEA-eng, 2025-02-28:
At least partially solved by #19913?

maxwxyz, 2025-02-28:
yes, reset is possible, I adjusted the original issue.

closed 2025-03-05 by chennes