FreeCAD/FreeCAD #19880: It is not possible to run two different versions in parallel on the same system (1.0 / 1.1)

maxwxyz, 2025-02-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

As seen in https://github.com/FreeCAD/FreeCAD/issues/19879 it is currently not possible to have independent versions of FreeCAD with different preferences and parameters on the same system.

Both, 1.0 and 1.1 use the same config files to store user preferences.
In the mentioned issue, Headlight is used for the default light in 1.0 and one of the three point lights in the new setup. The old direction is used in 1.1 which ignores the default position, making the changes not visible to the user and therefore not fixing the issues as intended.
If I adjust the Headlight, the same direction is used in the 1.0 version, making it impossible to use this version.

Maybe a solution could be to create a unique config file per FreeCAD version. 
Allow the first time a new FreeCAD version runs to migrate basic configs or to start fresh.
This way, I can have 1.0 and 1.1 on my system, without messing with each others preferences.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40421 (Git) Conda
Build type: Release
Branch: main
Hash: f9bb6508ffb420502d7b91592d87ea4f95c399d4
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/108.888
Installed mods: 
  * Cables-master 0.1.1
  * CfdOF 1.29.8
  * CurvedShapes 1.0.13
  * Curves 0.6.58
  * dodo 1.0.1
  * fasteners 0.5.37
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.1.1
  * parts_library
  * Rocket 4.1.1
  * sheetmetal 0.7.11
  * Silk 0.1.5
```

### Subproject(s) affected?

Core

### Anything else?

Maybe related: https://github.com/FreeCAD/FreeCAD/issues/18765

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

kaiwas, 2025-02-26:
There is a possibility that the differences between the versions will sooner or later have huge differences. I think we need to introduce an algorithm similar to the approach in Blender.

https://github.com/FreeCAD/FreeCAD/issues/16926

maxwxyz, 2025-02-26:
@kadet1090 mentioned a manual workaround:
you can run FreeCAD with FREECAD_USER_DATA overriden
`FREECAD_USER_DATA=~/.local/share/FreeCAD-dev ./bin/FreeCAD`

kaiwas, 2025-02-26:
> mentioned a manual workaround

If it's that easy - why isn't there a stable version without messing around with consoles and launch parameters?
(I find it easier to rename folders)

pixelwaster, 2025-02-27:
I have a batch file that gets copied into the extracted folder and use that to launch FreeCAD. To "uninstall" I just delete the folder.  Actually, I create a shortcut to the batch file, change the icon to the FreeCAD logo, and move the shortcut my desktop. The name is changed to march the release number. v1.0 and three weeklies all work without interfering with the others.

FreeCAD.bat -
`@echo off
set CURRENTDIR=%~dp0
set FREECAD_USER_HOME=%CURRENTDIR%
if not exist %CURRENTDIR%\cfg mkdir %CURRENTDIR%\cfg
start %CURRENTDIR%\bin\freecad.exe -u %CURRENTDIR%\cfg\user.cfg -s %CURRENTDIR%\cfg\system.cfg --write-log %1
`

szdowk, 2025-02-27:
Especially important, that 0.21.2 is still needed to open&work on old models due to incompatibility between 0.21 and 1.x.

kaiwas, 2025-02-27:
> FreeCAD.bat

Thank you! I know such solutions, but I think that an ordinary user should not know such workarounds and write launch rules. This is some kind of dark magic of programmers and system administrators. The user should just open the program and work in it.

The more global changes in the program accumulate - the more urgent this problem will become. And the sooner the improvement occurs at the program level - the easier it will be later.

furgo16, 2025-02-27:
I also think this feature request makes sense.

Just to add some additional data point to address `@kaiwas`' comment about users not having to use the command line to launch FreeCAD (which I agree with): a workaround is to use confined packages. The snap package already supports [parallel installs](https://github.com/FreeCAD/FreeCAD-snap/blob/master/docs/index.md#parallel-installs) which are completely independent from one another. With separate filesystem locations for configuration files. That's how I run both 1.0.0 and 1.1.0dev on the same machine. Flatpak is another confined packaging option, but I don't think it supports parallel installs.

This is not a solution to the issue though: it works only on Linux, and even there people might prefer other installation methods. Simply highlighting a workaround, as it's not a widely-known feature.

Nuck-TH, 2025-03-02:
> Thank you! I know such solutions, but I think that an ordinary user should not know such workarounds and write launch rules. This is some kind of dark magic of programmers and system administrators. The user should just open the program and work in it.

While i'm not against discussed feature, using nightly version already implies that you know good what you're doing and can deal with peculiarities and consequences(mainly incomplete features and intermittent bugs).

"Ordinary user" should stick to using stable release version.

maxwxyz, 2025-03-02:
Often we ask the original user who filed an issue to verify that it is fixed after a PR merges as not all issues can be reproduced on all systems. 
Also, when creating issues, users are expected to test it on the weekly version, if the issue in the stable version is still reproducible in the development version. So maybe "ordinary users" might not use the weekly daily but run it in parallel on the same system which might update some default parameters.

Nuck-TH, 2025-03-02:
Then it may be better not just use "separate location" for configuration, but make application self contained like bat file does, maybe by reacting to presence of file portable.ini like some other software does to do same thing, and ship nightlies with it included. This way behavior can be esiliy changed without additional commandline parameters, it won't by default affect release builds and users can both test nigtlies against release build, and multiple nightlies against each other without affecting each other via shared configuration.

chennes, 2025-09-03:
It's not *exactly* as you describe it, but this is basically implemented now, by #23321 (and some follow-up PRs fixing and polishing). There's no "start fresh" option, there is only "keep using the old directory" and "make a copy of the old directory for this version", but I think it does achieve the "run two versions in parallel" request that's fundamental here.

maxwxyz, 2025-09-03:
does "keep using the old directory" not create a v1-1 folder and use same configs?

luzpaz, 2025-10-03:
soft bug. what's the status of this ticket?

closed 2025-10-24 by Roy-043