FreeCAD/FreeCAD #19881: ArchDetail file broken in latest weekly

maxwxyz, 2025-02-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The ArchDetail.FCStd example file is broken if opened from the start page in the latest weekly.
There is no geometry visible and objects which are not recognized.
@yorikvanhavre FYI

[image]

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40421 (Git) Conda
Build type: Release
Branch: main
Hash: f9bb6508ffb420502d7b91592d87ea4f95c399d4
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/108.888
Installed mods: 
  * Cables-master 0.1.1
  * CfdOF 1.29.8
  * CurvedShapes 1.0.13
  * Curves 0.6.58
  * dodo 1.0.1
  * fasteners 0.5.37
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.1.1
  * parts_library
  * Rocket 4.1.1
  * sheetmetal 0.7.11
  * Silk 0.1.5
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-02-26:
It must be a regression since Monday as Git Rev 40408 (AppImage) is working fine.

[image]

```
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.40408 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: b07caa732e63064b8851ecdda184292962d95c15
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Logical/physical DPI: 96/93.1131
Installed mods: 
  * FreeCAD-themes 2025.1.7
  * DynamicData 2.71.0
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.29.8
  * fasteners 0.5.37
  * freecad.gears 1.3.0
  * Curves 0.6.58
```

Syres916, 2025-02-26:
In fact it's absolutely fine using the latest Git rev self compiled and there's been no changes since Monday on either Draft or BIM so I think could be something else in core that's been merged affecting Windows builds only???

```
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.40422 (Git)
Build type: Release
Branch: main
Hash: 56f0f640153926552effff9a3e931c8ce9a49e11
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, IfcOpenShell 0.8.1.post1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Logical/physical DPI: 96/93.1131
Installed mods: 
  * FreeCAD-themes 2025.1.7
  * DynamicData 2.71.0
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.29.8
  * fasteners 0.5.37
  * freecad.gears 1.3.0
  * Curves 0.6.58

```

maxwxyz, 2025-02-26:
@chennes @oursland any idea?

maxwxyz, 2025-02-27:
also confirmed on 
```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40421 (Git) Conda
Build type: Release
Branch: main
Hash: f9bb6508ffb420502d7b91592d87ea4f95c399d4
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/141.951
Installed mods: 
  * Curves 0.6.58
  * fasteners 0.5.37
  * freecad.gears 1.3.0
  * Manipulator 1.5.7

```

paullee0, 2025-02-28:
Seems nothing wrong here :)

OS: Fedora Linux 40 (Workstation Edition) (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.1.0dev.40408 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: b07caa732e63064b8851ecdda184292962d95c15
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/30.48
Installed mods: 
  * Link to SketchArchPlus

yorikvanhavre, 2025-02-28:
No problem on my side either (Debian Linux, today build), however I notice the TechDraw view suddenly has a wrong scale.. Maybe because of #19787 ?

[image]

maxwxyz, 2025-02-28:
No error message, no TechDraw page is visible. The above screenshots contain all objects.

yorikvanhavre, 2025-02-28:
It looks like no object were initialized successfully. That's really weird, because there should be errors, and it would indicate something not functioning in BIM or Draft. Does the BIM example model open correctly? Also do basic Draft/BIM stuff (Draft line, BIM wall...) work normally?

maxwxyz, 2025-02-28:
@yorikvanhavre BIM Example and a custom BIM file open without issues and problems, only the ArchDetail is corrupted.

maxwxyz, 2025-02-28:
If it helps, I just created a Draft Polyline in BIM and got this error:

[image]
Not sure if related to this or #19821 - I did not manually update anything.

furgo16, 2025-02-28:
> Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1

That's strange. From your About info it looks like you have IfcOpenShell installed (albeit that upstream build that erroneously reports its version, I think it's 0.7.0).

maxwxyz, 2025-03-01:
Still an issue in todays weekly.
@semhustej you are on Windows, can you confirm too?

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40444 (Git) Conda
Build type: Release
Branch: main
Hash: 089605bb4db98668661bf56db3f5f5bb4f7f0d87
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/108.888
Installed mods: 
  * Cables 0.1.3
  * CfdOF 1.29.8
  * CurvedShapes 1.0.13
  * Curves 0.6.58
  * dodo 1.0.1
  * fasteners 0.5.37
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.1.2
  * parts_library
  * Rocket 4.1.1
  * sheetmetal 0.7.20
  * Silk 0.1.5

```

Syres916, 2025-03-01:
If it helps someone who can bisect on Windows, using the last Windows 7 build available the last known good commit was https://github.com/FreeCAD/FreeCAD/commit/928556c5e525ad1fd4607aef613813ff80a8de36 on 21st February 2025 and with https://github.com/FreeCAD/FreeCAD/commit/f9bb6508ffb420502d7b91592d87ea4f95c399d4 being the first known bad so that's 105 commits between them.

semhustej, 2025-03-01:
I can confirm the issue. Tested also in safe mode. I get this error in the Report view:

```
17:00:47  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'OSError'>: Invalid project file C:/Users/semhu/Documents/tyo/BF/FreeCAD/FC_daily/FreeCAD_weekly-builds-40444-conda-Windows-x86_64-py311/data/examples/ArchDetail.FCStd: stoul argument out of range
```

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40444 (Git) Conda
Build type: Release
Branch: main
Hash: 089605bb4db98668661bf56db3f5f5bb4f7f0d87
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/92.6073

```

furgo16, 2025-03-01:
Other folks seem to have this error message with other file types as well: https://forum.freecad.org/viewtopic.php?t=95229

sclebo05, 2025-03-01:
I know that 'me too' doesn't help, but adding a datapoint. The ArchDetail file works in 40303(git) (with the many errors attached) 

[image]

The file shows Unnamed and 
"<class 'OSError'>: Invalid project file C:/Users/<username>/Downloads/FreeCAD_weekly-builds-40444-conda-Windows-x86_64-py311/data/examples/ArchDetail.FCStd: stoul argument out of range" in 40444 (git).

Syres916, 2025-03-01:
OK, with regard to what I can _prove_ only on Linux, please see https://forum.freecad.org/viewtopic.php?p=814249#p814249

So I'll try to create at least two new Unit Tests tomorrow with regard to https://github.com/FreeCAD/FreeCAD/pull/18676 and then we can see if one fails on the Windows CI builds and the other on the Linux builds, up to present no Mac users have reported these errors to the best of my knowledge.

sclebo05, 2025-03-01:
Yeah I wasn't clear that I was on Windows. Here's my About screen:

OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.40444 (Git) Conda
Build type: Release
Branch: main
Hash: 089605bb4db98668661bf56db3f5f5bb4f7f0d87
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/92.6073

Syres916, 2025-03-02:
The two Unit Tests are in https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:Unit_Tests_PR18676
**Under no circumstances** should this be used in core, it is purely to assist finding a solution, we need to create a test file from scratch in the near future. CC @kpemartin

kpemartin, 2025-03-02:
I haven't tested this yet (still building) but given the error I think I know what is causing this, or at least I can think of a situation where the new unique-name generation could get such an error.
If there is a name in the file which has (what appear to be) uniquefying digits whose value exceeds the upper bound of an unsigned long this might occur. The old unique-name generation kept the unique values as strings and did "string arithmetic" to work with them and so had no upper bound on the number of digits or their value when interpreted as an integer.

If this is the problem first I'll add a unit test, then go about fixing it. Does anyone know if there is an arbitrary-precision integer arithmetic package available?

kpemartin, 2025-03-02:
Problem reproduced on Windows. The file contains an object names `Compound006002002002002` and the UniqueNameManager fails when trying to parse those digits as an unsigned long. Such names original in older versions of FC which just added 3 unique digits to a colliding name without first checking if it already have uniquifying digits that could instead just be incremented.

I've just created a Draft PR #19943 which will address this issue. So far the PR only contains a new unit test (which fails)

kpemartin, 2025-03-02:
I'm not sure why this succeeds on non-Windows platforms. Perhaps std::stoul returns a nonsense value, rather than throwing an exception, when given the digits `006002002002002`. This would show in an apparently successful open of the file as an object somewhere in the model called "Compoundnnnnn" where the n's are some other number, likely fairly large.

furgo16, 2025-03-02:
I can see such a compound here, using Linux:
[image]

There's also a few more.

Syres916, 2025-03-02:
> I'm not sure why this succeeds on non-Windows platforms.

What's far more concerning for me is that the Windows Conda build that users are provided errors when loading _ArchDetail.FCStd_ but the two CI Windows Builds passed without issue, if anything it should be that the CI builds are more or at least as stringent than the user issued build(s). FYI @adrianinsaval

maxwxyz, 2025-03-02:
@oursland FYI

kpemartin, 2025-03-02:
Actually, these names are all coming from an existing file and so are all unique already. My conjecture about them having unexpected numbering revolves around them being newly-generated unique names.

The error in std::stoul is happening when the name is registered with the unique name manager (to avoid collision with future duplicates).
On Windows `std:;stoul("006002002002002")` causes an exception. It turns out that the reason is that (unsigned long) is 4 bytes, at least in the Windows environment, so 6002002002002 is too large a value.

In other environments with longer longs it would quietly return the correct value, but then the calling code would stuff it into an unsigned int, truncating it modulo the size of that type. In this case, other than unexpected values for new unique names, the only effect would be false collisions (e.g. between name001 and name4294967297).

Even in such environments, though, a much longer digit string could still cause the same exception.

I've just pushed a fix which involves adding some limited arbitrary-precision unsigned integral arithmetic. I still have to review the lint and make some code changes (ref arguments, copy ctors, encapsulate magic constants, etc) so it doesn't run like a dog.

kpemartin, 2025-03-03:
@aapo-aapo That is indeed the point of the arbitrary precision arithmetic being added. The new code in theory could handle a number with a value up to (2^32)^(2^32), or perhaps more if the size on std::vector were more than 32 bits. In this implementation, though, to make conversion to/from digit strings much simpler, the limit would instead be 1000000000^(2^32), but that is still about a 36-billion-digit number. Of course, memory limitations would put a much smaller practical limit on such numbers as well.

closed 2025-03-31 by chennes