FreeCAD/FreeCAD #19882: View goes to nowhere

drwho495, 2025-02-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When padding a sketch when the draft plane is enabled in part design, the view flies to nowhere. Same happens when you open a sketch, this behavior will continue even after you disable the draft plane, but does not happen before you enable the draft plane

Steps to reproduce:
 1. Create a new document
 2. add a body
 3. Create a sketch in the body
 4. Add a rectangle in the sketch
 5. Enable the draft plane
 6. Close the sketch and pad it

Grid settings:
[image]

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (GNOME/gnome-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40408 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: b07caa732e63064b8851ecdda184292962d95c15
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Logical/physical DPI: 96/161.377
Installed mods: 
  * Render 2024.12.15
  * ArchTextures
  * CadbaseLibrary 2.0.0
  * Behave-Dark-Colors 0.1.1
  * SearchBar 1.0.2
  * WebTools 1.0.0
  * Manipulator 1.5.7
  * POV-Ray-Rendering
  * Ondsel-Lens 2024.11.29.01 (Disabled)
  * freecad.gears 1.3.0
  * ExplodedAssembly
  * fasteners 0.5.37
  * FreeCAD-Telemetry 1.0.0beta
  * QuickMeasure 2022.10.28
  * FreeCAD-Ribbon 1.7.3.7
  * assembly2
  * Freecad-Built-in-themes-beta 1.2.2
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

drwho495, 2025-02-26:
Occurs in 40077 aswell

maxwxyz, 2025-02-27:
There is a property which you can set to false

[image]

drwho495, 2025-02-27:
oh thanks

> There is a property which you can set to false
> 
> [image]

drwho495, 2025-02-27:
that doesnt seem to fix the issue, the issue is when opening the sketch or creating a pad

maxwxyz, 2025-02-27:
Could you share a file and screencast?

drwho495, 2025-02-27:
Now I can't get the behavior to repeat, but here are files of an effected project. Open the first sketch in the file called passive claw.

[PassiveClaw.zip](https://github.com/user-attachments/files/19015345/PassiveClaw.zip)

maxwxyz, 2025-02-27:
Try to reset your orbit center (middle mouse button) on the geometry. It was way off. Then all is fine

drwho495, 2025-02-27:
> Try to reset your orbit center (middle mouse button) on the geometry. It was way off. Then all is fine

oh that fixed it, thanks!