FreeCAD/FreeCAD #19883: TechDraw: Problems with the section types of the part

taulog, 2025-02-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

(I'm sorry for any inaccuracies in the presentation, English is not my native language, I use machine translation.)
___
1) It is impossible to realize a full-fledged view of the section of the part if an incision enters the thread. If the incision is removed, then the cross section is realized.
2) The Crosshatch is not always displayed on the cross section (I did not understand what it depends on).
3) For a very long time (about 2 min.), and with great difficulty for the computer, axial lines (between two lines or points) are created in the drawing.
4) Unfortunately, there is no way (at least I did not find it) to depict the carving in a simplified form in the drawing. This would save resources and make the drawing more correct (for Russia at least). The last photo shows how it is depicted in a real drawing. I would really like this feature to appear in your program.
(I use the "FreeCAD Fasteners Workbench" to create threads)
[zzz.zip](https://github.com/user-attachments/files/18994012/zzz.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: unset/FreeCAD Light/Fusion
```

### Subproject(s) affected?

None

### Anything else?

[image]
[image]
[image]

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-04-11:
@taulog can you test on the development[^1] version and report back ?

[^1]: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

FEA-eng, 2025-09-30:
Seems to be still relevant (at least the lack of hatch)

@WandererFan FYI

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```