FreeCAD/FreeCAD #19887: HiDPI / Retina font size differences

JohnOCFII, 2025-02-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Creating this issue related to a recent Discord discussion with @hyarion 

Default text font size on macOS is noticeably smaller on MacOS (at least on my 16" MacBook Pro) when running macOS vs. running Debian Linux in a VMware VM.  Both of these screen shots are from Safe Mode.  Both have Sketcher text size of 17px.

You can see that the default DPI is different in each environment.  Just posting the information here -- not sure what better way there might be to keep text sizes similar, so that users don't have to increase font sizes as often on macOS, or deal with other user interface elements that often appear much smaller on macOS than on Linux.

### Full version info

```shell
Linux

OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome/xcb)
Architecture: arm64
Version: 1.1.0dev.40317 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: b67ab979afdba808fc03196faf9c5f8830df471f
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/96.1473

========

Mac

OS: macOS 15.3.1
Architecture: arm64
Version: 1.1.0dev.40419 (Git) Conda
Build type: Release
Branch: main
Hash: cf477695f9662d9892d5fc79e83065b4ac3d52bf
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell /.0.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 72/127.5
```

### Subproject(s) affected?

None

### Anything else?

macOS image:

[image]

Debian Linux image:

[image]

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

oursland, 2025-02-27:
I experience this as well.  It becomes extremely apparent when I am using FreeCAD on an external monitor and drag it to the MBP display.

tringenbach, 2025-02-27:
Is this issue intended to be fonts only? There are other differences as well (present in your screenshot):

- the navigation cube (upper right) is half the size on Mac
- the widths of the lines of the sketch are narrowing on the Mac
  - (Not in the screenshot, but, I have the background grid turned on normally, and I find the thinner lines start to blend in to the grid)
- the red constraints are smaller too (which is both text and icons)

JohnOCFII, 2025-02-27:
> Is this issue intended to be fonts only? There are other differences as well (present in your screenshot):

My initial concern was fonts, but looking more closely, there are other differences (as you and others have noted) as well.

hyarion, 2025-02-27:
Wmware seems to report the wrong screen size. Look at the values for Logical/physical DPI.

Does anyone have a mac which can be dual booted to check how this works natively?

JohnOCFII, 2025-02-27:
> Wmware seems to report the wrong screen size. Look at the values for Logical/physical DPI.

That may be true, but I think the primary point is that many user interface elements are, by default, quite small on macOS.  (And yes, while I'm no longer young, with glasses, my vision is correct to 20/20 near and far.)

tringenbach, 2025-02-27:
Here's a screenshot from my 2019 16" MacNook Pro. I tried to recreate what you did as for as the sketch. I picked light theme. And I'm running off of git / pixi with qt6.

```
OS: macOS Sequoia (15.3)
Architecture: x86_64
Version: 1.1.0dev.40408 (Git) Conda
Build type: Release
Branch: main
Hash: b07caa732e63064b8851ecdda184292962d95c15
Python 3.13.0, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical/physical DPI: 72/132.25
```

[image]

hyarion, 2025-02-27:
@JohnOCFII what I'm wondering is if this is a mac issue or all system have this issue.

JohnOCFII, 2025-02-28:
I've been looking for an article that describes how Apple gets to the "default" Retina resolution for the (newer) MacBook Pros).  These two together are a pretty good start.  

https://9to5mac.com/2021/10/19/new-macbook-pro-screen-resolution-options/

https://havn.blog/2024/11/14/display-resolution-guide.html

So this issue might be specific to the MacBooks, in terms of how the move between physical and logical resolution.

hyarion, 2025-02-28:
Qt should already handle this. We already see the difference her between logical and physical dpi.

The questions are:
* why is wmware reporting this wrong?
* will windows and Linux look the same when they have the same dpi settings?

JohnOCFII, 2025-02-28:
Interesting questions.

Within VMware you can change the display settings.  The information as above was with the default settings.  I switched it to use the "full" Retina resolution, and as expected, it displays at a higher resolution (much higher than the default Mac Retina resolution, it would appear) thus, everything is very tiny on the screen.  Note that the reported DPI from FreeCAD did not change.

```
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome/xcb)
Architecture: arm64
Version: 1.1.0dev.40317 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: b67ab979afdba808fc03196faf9c5f8830df471f
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/96.063
```

[image]

[image]

oursland, 2025-02-28:
For clarification, you found that some parts of the UI such as NaviCube were not multiplying their dimensions against `devicePixelRatio`?

tringenbach, 2025-02-28:
> For clarification, you found that some parts of the UI such as NaviCube were not multiplying their dimensions against `devicePixelRatio`?

NaviCube is the only part looked at in any depth, and it is not multiplying its dimensions against `devicePixelRatio`. So for other places, I am just guessing that they have the same problem. 

So my plan is to make a PR that only fixes NaviCube, and learn from whatever feedback I get from that before trying to fix other places.

oursland, 2025-02-28:
It's wild how the whole codebase needs to be sprinkled with `* devicePixelRatio()`, which is not trivial to get for all classes.  For example, in the `NaviCube` example, one gets this via:

`scale = info.tabWidget->titleBar->grab().devicePixelRatio();`

It seems to me that one should be able to configure the renderer to account for the DPI and one simply render in the desired screen dimensions.

@kadet1090 You have more understanding than me about the rendering system.  Is there a way we can avoid all of these calls to `devicePixelRatio()` in every component that wants to render?

tringenbach, 2025-02-28:
@oursland Currently, I am grabbing it with `m_View3DInventorViewer->devicePixelRatio();` in `NaviCube`. I haven't check if that is correct or not though, and I haven't figured out how to get hook into the event for when it changes yet either. (The window can get dragged to another monitor with a different ratio).

I agree that this whack-a-mole approach isn't good.

I did see that `src/Gui/Quarter/QuarterWidget.cpp` is making use of `devicePixelRatio`.  That seems important, that suggests that Coin is at least somewhat aware of it too, but I don't really know what's going on there yet.

hyarion, 2025-03-01:
> Is there a way we can avoid all of these calls to devicePixelRatio() in every component that wants to render?

For OpenGL, can't we just adjust the projection matrix to compensate for the difference between logical and physical pixel sizes?

oursland, 2025-03-01:
> For OpenGL, can't we just adjust the projection matrix to compensate for the difference between logical and physical pixel sizes?

That's what I thought as well.

tringenbach, 2025-03-01:
I am not really sure how to adjust the projection matrix.  I also don't quite have my head around if that makes sense.  I guess it's sort of like oversampling (except the screen actually has that many pixels, so there's no downsampling step). But a lot of things are drawn at a fixed size regardless of where the camera is pointing.

Anyway, I dug through the code trying to understand it a bit better and looking for the right place to adjust a projection matrix. I wasn't successful in the latter, but I did find this bit of code in the Sketcher: https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Sketcher/Gui/EditModeCoinManager.cpp#L369-L398 which links to some forum threads.

I think the approach taken there is wrong, at least in 2025, because it tries to use the exact pixel ratio. However, for Macs on QT, there's instead a fixed scaling number (2.0 in my case).  (I also didn't verify if the code was correctly doing what it was trying to do, I kinda suspect not just because of how small things look.)

So just as a test, I hard coded the values to "2.0f". I also multiplied by 2 the `constraintIconSize` and `markersize`. Here's a screenshot of that (which also still includes the change to NaviCube):

[image]

tringenbach, 2025-03-02:
I posted a PR for the NaviCube changed I mentioned: https://github.com/FreeCAD/FreeCAD/pull/19929

With those changes, on my MacBook Pro, it now looks like this:

[image]

If I change my Mac's settings from defaul tto the max resolution, it instead looks like this:

[image]

And the about dialog reports this:
```
OS: macOS Sequoia (15.3)
Architecture: x86_64
Version: 1.1.0dev.40409 (Git) Conda
Build type: Release
Branch: feature/hidpi-navicube-fix
Hash: 2ce1b51b8fd2e8c80ee0082a0874b40255cb9013
Python 3.13.0, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical/physical DPI: 72/302

```

I'm surprised it says 72/302, maybe I don't understand what that is telling me.

bombela, 2025-03-02:
Fresh install, deleted user config. The fonts are so tiny, it is almost impossible to read even standing 1.5m away from a 65" 4k monitor. The icons are bigger then pre-freecad 1.0 interestingly enough. Though previous versions (and the linkstage fork from RealThunder) had no issue with the font size.

[image]

hyarion, 2025-03-02:
@bombela can you copy your full info (help -> about -> copy to clipboard) and paste it here?

tringenbach, 2025-03-06:
@JohnOCFII @oursland  I think, based on reading the links I posted above on this issue, that it's important to know the devicePixelRatio for this discussion. So I opened an PR to add it to that line.

https://github.com/FreeCAD/FreeCAD/pull/20013

tringenbach, 2025-03-07:
I opened a draft PR for the Sketch changes I mentioned earlier: https://github.com/FreeCAD/FreeCAD/pull/20038

Here's what it looks like for me:
[image]

```
OS: macOS Sequoia (15.3)
Architecture: x86_64
Version: 1.1.0dev.40584 (Git) Conda
Build date: 2025/03/07 03:03:43
Build type: Release
Branch: feature/sketcher-device-pixel-ratio
Hash: c37b48e81f2f7da46ec88c5bacadc57f21c3de8e
Python 3.13.0, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 72/132.25/2

```

tringenbach, 2025-03-07:
@hyarion and @oursland:

> For OpenGL, can't we just adjust the projection matrix to compensate for the difference between logical and physical pixel sizes?

I guess I'm not getting how that would work. I haven't tried it yet, I'm not really sure where in the code I would adjust it.

But thinking through it, wouldn't that result in rendering the font tiny and then scaling it up? NaviCube is see is rendering texturing. I'm not sure how Sketcher works. (I've only looked at the code that sets the font sizes so far, I haven't found how it draws the text and lines  yet.) But I would think it would have the same problem. Something has to know to render the font at `32px` instead of `16px`, or else you're only going to have the font drawn at 16px and then scaled up.

But maybe it's doing something fancier than I realize. I know if you want really good text with subpixel rendering, you need to know exactly what kind of screen you're rendering to and all this stuff, and I'm not sure how that works with OpenGL or any kind of perspective matrix, unless the font render is running as a shader or something like that and takes it all into account.

Anyway, that's why I'm skeptical. But I'm also happy to be wrong.

hyarion, 2025-03-07:
> But thinking through it, wouldn't that result in rendering the font tiny and then scaling it up?

You are correct, usually what people do is to adjust textures that are size dependent to the same amount to avoid them to look blocky (or lower dpi).
Line/Point attributes (like line width) also needs adjustments as they are defined in viewport space.

The benefit of doing it in this general fashion is that only a few places needs to be adjusted, instead of everywhere.
Another good thing is that it either works or doesn't work, we don't get into a situation that only most of the elements are adjusted for HiDPI, while other aren't.

depthoffocus, 2025-05-10:
This is 41628, macOS x86, on a 13" MacBook Pro at its default resolution (1440x900, 2x). 

I have set line widths I appear to be able to set, to 1px. I have the marker sizes set to 5px, font size set to 12px and view scale set to 1.00.

I think broadly the change to having, device independent, CSS-style pixels and font sizes has worked really well. This always felt a bit inconsistent, and the font sizes Mac users had to choose didn't seem to be logically tied to anything.

But I think some guidance might be needed:

- What sets the thickness of the axis lines? These feel chunkier than they need to be.

- Is "view scale ratio" now ignored?

- What sets the thickness of the (red) length/distance constraint indicator lines etc.?

- The shape preview lines when drawing now seem _excessively_ thick (this gets in the way of the dimension preview) and I can't see anything that changes this? The rubber-band selection lines are fine.

- Perhaps the constraint marker icons feel a bit large still.

Most of the line sizes etc. feel slightly more comfortable if I boost the display resolution to 1680x1050, which is the higher resolution mode supported by the 13" MacBook. But it's not the default; these machines default to 1440x900, which is 1.77x scaled.

Thank you for working on this -- it really will make a difference to the Mac experience.

https://github.com/user-attachments/assets/9c76e90f-6ad8-4faf-8e0a-93a47816c955

hyarion, 2025-05-10:
@depthoffocus there's a lot to unpack here, but I think the first question is: has the HiDPI issue been fixed? @JohnOCFII what do you think?

When we establish that the HiDPI issues are fixed then we can look at adjustments. Otherwise we might fix it here, but cause issues on other systems.

depthoffocus, 2025-05-10:
> @depthoffocus there's a lot to unpack here, but I think the first question is: has the HiDPI issue been fixed? 

I didn't want to get discouraging because I really appreciate the work on something so thorny, but since you ask this way, I think the answer is no.

To me, a fix to HighDPI would be having more or less the same flexibility, but with DPI-independent numbers.

The core question of sketch lines and the constraint text, yes, these two elements are now sized sanely. I do think that the result of setting 1px for sketch lines is right-sized, and the text font size is appropriate for the pixel size you set.

But a lot else now feels substantially over-large. On my Mac at least (a pretty common configuration) it feels like it has now become disproportionately chunkier. 

Perhaps it is just a question of ensuring the same change is applied to these other elements.

(I hadn't checked grid lines before I commented; they seem to still be operating under the old scale.)

It is possible I have not seen the settings that affect these things. And I will try to do a comparison with a typical low DPI Windows setup at the equivalent 1x resolution. Probably with VirtualBox.

I think a bigger problem in a way is that 1px in Sketcher now produces lines that are about as thick as 2px in the settings that control line thickness for 3D shapes, but I understand that this can only be addressed in this progressive way.

hyarion, 2025-05-10:
I appreciate the feedback you are giving. The problem here is that it sounds like everything works as intended.

That said, you do have good points. I also felt this myself, but shrugged it off as something I would get used to due to have getting used to the buggy behavior for years.

You are correct that there's currently a difference between sketcher and other elements' thickness in the viewer and that's because this hasn't been applied everywhere.

Back to why this happens;
There's physical pixels and logical pixels and the fix for this issue was to ensure logical pixels are used instead of the previous solution.

It might be that some sizes, like line width, should be defined in physical rather than logical pixels.

There is some work in progress to redefine how sketch should look like by default and maybe this need to be part of that work.

I'm not dismissing your feedback here, it is valid, but I'm wondering if we should to move it to a new issue and close this one as this one was to make sure it looks the same - and you, rightfully point out - that maybe we don't want the *exact* look.

What do you think?

JohnOCFII, 2025-05-10:
@hyarion 
> there's a lot to unpack here, but I think the first question is: has the HiDPI issue been fixed? [@JohnOCFII](https://github.com/JohnOCFII) what do you think?
>
> When we establish that the HiDPI [...]

I had my first look at the changes last night.  DEFINITELY NOTICEABLE!  :) It feels like when I moved to test on Linux vs. Mac.  Perhaps a little too heavy with the lines now, but as you say, that can be tweaked.  Some of the items (marker size, text size) that I had manually increased, I then decreased.  Would love to see these exchanges extended bit by bit.  Say, to Part and PartDesign next.

Before:

[image]

After:

[image]

After closing Sketcher and back to PartDesign: (thin lines again)

[image]

depthoffocus, 2025-05-10:
> I'm not dismissing your feedback here, it is valid, but I'm wondering if we should to move it to a new issue and close this one as this one was to make sure it looks the same - and you, rightfully point out - that maybe we don't want the *exact* look.
> 
> What do you think?

Well, I didn't raise the issue so it's not my call, I think. I am not really communicating my thoughts very well, so I will leave this to others who have more experience of FreeCAD away from the Mac. :-)

But if the intended outcome of this issue was to get to "logical" DPI-independent pixels for all the existing pixel-based preferences then it does at least seem incomplete, because these scale changes don't seem to have been applied to the pixel width preferences for the grid lines. Those appear to still be device pixels.

hyarion, 2025-05-10:
> Well, I didn't raise the issue so it's not my call, I think. I am not really communicating my thoughts very well, so I will leave this to others who have more experience of FreeCAD away from the Mac. :-)

Everyone is free to comment :)

> But if the intended outcome of this issue was to get to "logical" DPI-independent pixels for all the existing pixel-based preferences then it does at least seem incomplete, because these scale [...]

That is a good point 👍

hyarion, 2025-05-12:
@jacquesbeaurain thanks for you help, it would be awesome if you'd like to work on this too!

I think it would be good to discuss this work with CWG (cad working group) and DWG (design working group). They mostly communicate through [discord](https://discord.com/invite/w2cTKGzccC) so join in there if you like.

I'll ping @pierreporte and @kadet1090 as they are part of those groups.