FreeCAD/FreeCAD #19888: The Del Key dont work in Sketcher (FreeCAD 1.0)

wadysal, 2025-02-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I'm editing a Sketch, select the objet to delete and press the Del key and nothing happen. Any suggestion? Thank's a Lot

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Spanish/Spain (es_ES)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * InventorLoader 1.5.1
  * sheetmetal 0.7.11
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-27:
Please provide an example file and steps how to reproduce. It works here.

luzpaz, 2025-03-15:
No response from OP. Closing.

closed 2025-03-15 by luzpaz