FreeCAD/FreeCAD #19893: Part: A model with chamfer or fillet created in part module of FC 0.21.2 cannot be shown in other mode than wireframe in FC 1.x

szdowk, 2025-02-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Possible incompatibility issue. A model (e.g. box) with chamfer or fillet created in Part module of FC 0.21.2 cannot be shown in other mode than wireframe in FC 1.x. 
Recompute, re-selecting edges/faces or changing the size of that chamfer or fillet in FC 1.x do not help. The "workaround" is to delete the chamfer in FC 1.x and create it again.

To reproduce:
1. Open FC 0.21.2, go to Part module.
2. Create a basic box or cylinder.
3. Use option "chamfer" or "fillet", select a face or edge, create chamfer or fillet.
4. Save.
5. Exit from FC 0.21.2, start FC 1.x and open saved model.

### Full version info

```shell
OS: Slackware 15.0 x86_64 (KDE/xcb)
Architecture: x86_64
Version: 1.1.0dev.40408 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: b07caa732e63064b8851ecdda184292962d95c15
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/81.5973
Installed mods: 
  * sheetmetal 0.6.13
  * Help 1.0.3
  * freecad.gears 1.3.0
  * Curves 0.6.53
```

### Subproject(s) affected?

Part

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-27:
Please add an example file and screenshots of the issue in the original post.

Syres916, 2025-02-27:
@maxwxyz this is in effect a duplicate of the already pronounced Blocker https://github.com/FreeCAD/FreeCAD/issues/18584 and there was a proposed solution but apparently a migration policy was being looked into but I've not seen any feedback regarding that.

szdowk, 2025-02-27:
@maxwxyz Yes, it is!

closed 2025-02-27 by maxwxyz