FreeCAD/FreeCAD #19894: Show placement helper visibility issues

maxwxyz, 2025-02-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The new show placement property (https://github.com/FreeCAD/FreeCAD/pull/19671) currently has three issues:
- it is always visible, even when the object is hidden:

[image]

- it is not visible if it is behind geometry:

[image]

- it sometimes displays multiple placement axes at different locations for one body:

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40421 (Git) Conda
Build type: Release
Branch: main
Hash: f9bb6508ffb420502d7b91592d87ea4f95c399d4
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/141.951
Installed mods: 
  * Curves 0.6.58
  * fasteners 0.5.37
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-28:
@kadet1090 the placement is also shown in wrong orientation for solids of imported step objects. The transform origin at object origin is correct:

[image]

kadet1090, 2025-04-02:
I'm not able to reproduce the 3. point or the one about step files, can you share me your files?

maxwxyz, 2025-04-03:
@kadet1090 

[demo.zip](https://github.com/user-attachments/files/19587041/demo.zip)

kadet1090, 2025-04-03:
Cannot really repruduce with your file on my branch:

[image]

However I found another issue, for some reason there are way too many indicators! I fixed most of them, but primitives inside do display their own placement indicators and this is probably what you misinterpreted. I'll fix that too and then everything should be ready to go.

kadet1090, 2025-04-04:
Ah ok, I misread it. Then it is fixed - but if you have file that is saved with multiple indicators visible it will need manual fixing.

kaiwas, 2025-04-08:
@maxwxyz As you understand - this is a sore subject for me :)
My issue is not only about the problem of visibility behind the object.
If my thoughts also turn out to be useful, that's good. The problem is really very important. Thank you!

maxwxyz, 2025-04-15:
@kadet1090 as I've understood it will not be fixed for existing models but correct for new. In the latest weekly it works as expected but it still is visible even when the object is hidden. Also, Part objects do not have this property.

kaiwas, 2025-04-15:
I don't know if this issue has been reported, but I noticed that the helper is not displayed for the expanded array elements.
Should I create a separate issue for this?
[image]

kaiwas, 2025-04-15:
I think I got it. Something happens when changing the profile based on the sketch (my development that I am working on).
Probably the problem is quite narrow and should not be paid attention to. I showed the procedure in the video.

https://github.com/user-attachments/assets/ede51e44-f843-4509-a785-42cf7ba11418

maxwxyz, 2025-06-19:
Still an issue: helper stays visible even if object is hidden.

maxwxyz, 2025-06-26:
to add here: also the show plane property stays visible if the sketch is hidden.

karamellpelle, 2025-06-26:
> Still an issue: helper stays visible even if object is hidden.

> to add here: also the show plane property stays visible if the sketch is hidden.

I.e. the fix in PR #20603 is not working (https://github.com/FreeCAD/FreeCAD/pull/20603#issuecomment-2961642704).

pierreporte, 2025-06-26:
I don’t think why it should be visible if the object is hidden. And an easy workaround could be to hide every children of the object.

kadet1090, 2025-06-26:
Placement Indicator and Plane Indicator are not children of object, simply annotations (additional visualisations) attached to object and as of now these are always displayed. Changing that would either require addition of yet another annotation type (that is hidden with object), doing hacks or changing it globally which can cause problems for other types.

If we want it for placement indicator only I think that I found why solution in #20603 is not working anymore so I can submit a patch.

karamellpelle, 2025-06-26:
Well, I was going to create an issue since I considered this a bug ( then found this issue), because:

* I didn't understand those random axes-objects in the 3D view (but later found out that they were placement indicators of hidden objects)
* I (temporary) hide objects because they "visually pollutes" my work. If placements are still showing then they continue to pollute.
* Hiding the placement indicator if object is hidden is probably how most 3D modelling software behaves (at least in Blender: _Properties > Object > Viewport Display > Show > Axes_).

Here is a suggested solution if you want to interact with placements instead of the actual object: https://github.com/FreeCAD/FreeCAD/issues/19894#issuecomment-2961526964

luzpaz, 2025-12-29:
soft bump

PR #20603 (2025-04-13): Gui: Show Placement Indicator - several fixes
This PR fixes #19894:
1. Indicator is now visible only when object is visible
2. Indicator is now shown on top of the object
3. Indicator is not not available for features of body as body is one object.