FreeCAD/FreeCAD #19902: Inconsistent placement and appearance of the ok button

boreq, 2025-02-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When setting a constraint using a varset the second window which opens has an OK button which is placed on the opposite say of the "negative action" e.g. clear/cancel, doesn't have a little icon and both buttons are aligned to the other side of the dialog window. This sometimes makes me click the wrong button.

[image]
[image]

### Full version info

```shell
OS: Arch Linux (hyprland/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.13.2, Qt 6.8.2, Coin 4.0.3, Vtk 9.4.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * fasteners 0.5.37
  * Curves 0.6.58
  * Gridfinity 0.9.7
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-27:
@FreeCAD/design-working-group FYI
Related: https://github.com/FreeCAD/FreeCAD/issues/13120

benj5378, 2025-02-27:
How old are these screenshot/FreeCAD version? Please check the newest available version (weekly build)

boreq, 2025-02-28:
> How old are these screenshot/FreeCAD version?

Latest release, 1.0.0, took them just before copying the version info so they aren't old screenshots.

> Please check the newest available version (weekly build)

I'm struggling to get those working I'm afraid, getting qt errors, I will report back if I manage to get `main` compiled instead.

boreq, 2025-02-28:
Ok, I manged to get the weekly build running. This is what those look like in that latest weekly build. I'd say that the situation is identical.

[image]
[image]

closed 2025-06-04 by chennes