FreeCAD/FreeCAD #19905: Floating toolbar for functions that are common to workbenches working with 3d space

belinski-roman, 2025-02-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Many CAD systems have a similar interface solution and it is often justified. An important note is that the panel should be a direct part of the 3d window and always located at the top and in the center (it will be absent in other workbenches that do not have a relation). Below are examples from other design systems.

As a rule, this panel contains tools for enabling a specific orthogonal view, aligning the view relative to the selected plane, elements for adding auxiliary geometry to the construction, and other things that are often needed during the modeling process.
Good candidates in FreeCAD for floating view are panels such as Structure and View. In my opinion, it makes sense to add a button to launch Persistent Section Cutting (although perhaps some more buttons will be appropriate there).  Here's a rough layout of how I imagine it. 

### Full version info

```shell
OS: Windows 10 build 19044
Architecture: x86_64
Version: 1.1.0dev.40421 (Git) Conda
Build type: Release
Branch: main
Hash: f9bb6508ffb420502d7b91592d87ea4f95c399d4
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/102.299
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Mannshoch, 2025-02-28:
I like the idea of such a toolbar. I think this panel should only contain viewing, selection helping or get information tools.
Feature in this panel should be usable while other feature are open. At least they pause a feature while used and resume it on finish. e.g. use a [clipping plane/section cut](https://forum.freecad.org/viewtopic.php?t=89160#p771295) to look into a body, then use measure and get a value you wish to use in a feature you had open the moment before you used clipping plane.

For the record, I'm not against manually add other features. But FreeCADs "default" should be a Features that follow this strict rule. So every body could click anytime on this toolbar without worry something persistent happens.

maxwxyz, 2025-02-28:
@FreeCAD/design-working-group is this a relevant feature or not in line with general UX roadmap?

Pendelf, 2025-02-28:
Great idea and it looks good. I think it's worth discussing in the appropriate discord thread

kaiwas, 2025-02-28:
In general, the idea of ​​filling the 3D window with useful things is very good. Now it is just an area for viewing the model. A lot of necessary elements could be brought to the 3D window to speed up the modeling process. If this is not on the roadmap, it is bad news.

pierreporte, 2025-02-28:
There are features that could benefit from this already, and some proposed features would need it, like isolation mode for assemblies.

belinski-roman, 2025-03-01:
> There are features that could benefit from this already, and some proposed features would need it, like isolation mode for assemblies.

Is the isolation function that you mentioned discussed somewhere? It's just that this is the first time I've heard of it, but I'm sure I'd like to be able to temporarily hide everything except the explicitly specified objects.

pierreporte, 2025-03-01:
@belinski-roman It’s discussed in ##12142. Somehow I forgot to indicate the issue number.

pierreporte, 2025-03-01:
I think that I’m mistaking this issue with #5563.

Mannshoch, 2025-03-02:
Another propose, I would preset this Toolbar Workbench/Addon independant.
A Workbench/addon could:
- Add own icon
- Deactivate the toolbar
- Replace the whole toolbar

Advice for Developer of Workbench/addons:
"please do not remove single icons, we would like to give users an as consistent experience as possible."

e.g. This would give #5563 the possibillity to replace this toolbar temporary with a flight control.

kaiwas, 2025-04-15:
One panel could have a fixed set of functions. But it would also be possible to make a "custom" floating panel with user tools or sets of standard actions that the user himself adds to the panel.

hyarion, 2026-02-07:
Why not just make this floating area a place where users can dock any toolbar groups? That way we wouldn't need to reinvent toolbars and allow users 100% control.

belinski-roman, 2026-02-07:
@kadet1090 what do you think about the implementation via the Python API, is it acceptable?

Mannshoch, 2026-02-13:
that sound nice.

- I see an anchor for docking any kind of toolbar as floating toolbar
- This docking anchor could have a possibility to be [temporary replaced](https://github.com/FreeCAD/FreeCAD/issues/19905#issuecomment-2692621237)
- The default dock from a FreeCAD standard Setup should contain a toolbar that only contain features that could be used imediately and every time and do not interfere with other features like a currently open sketch or any other currently open Feature. (simply change a view, appearing or give a report)

ArturWroblewski, 2026-04-17:
It looks great, that's what I was missing.

Could you also create prefabricated toolbars with names like:

SolidWorks Floating Toolbar - Filled, imitating the Solid Works icon set
Catia Floating Toolbar - Filled
3DEXPERIENCE Floating Toolbar - Filled
Autodesk Inventor - EMPTY

And when you select a navigation style, the toolbar in the floating menu/toolbar space would automatically activate.

[image]