FreeCAD/FreeCAD #19906: FC 40317 crash to desktop- While performing a boolean cut.

JDMils, 2025-02-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I am trying to cut a section out of a part on a disk as per my forum post here:
https://forum.freecad.org/viewtopic.php?p=813916#p813916

When performing the cut, FC crashes to desktop.

To reproduce the issue
* Select the Body.
* Control-select the Sketch_On_Surface.
* Open the Part workbench.
* Click Part | Boolean | Cut.
* Message shows "The use of non-solids for boolean operations may lead to unexpected results..."
* Click "Yes" to continue.
* FC crashes to desktop.

Windows Event Viewer shows:
`Log Name:      Application
Source:        Application Error
Date:          28/02/2025 8:35:31 AM
Event ID:      1000
Task Category: Application Crashing Events
Level:         Error
Keywords:      
User:          JDMILS02\julian
Computer:      JDMILS02
Description:
Faulting application name: freecad.exe, version: 1.1.0.0, time stamp: 0x67b8deca
Faulting module name: ntdll.dll, version: 10.0.26100.3037, time stamp: 0x95e6c489
Exception code: 0xc0000005
Fault offset: 0x0000000000163ea7
Faulting process id: 0x3DD4
Faulting application start time: 0x1DB895E769D7E66
Faulting application path: C:\Program Files\FreeCAD_weekly-builds-40317-conda-Windows-x86_64-py311\bin\freecad.exe
Faulting module path: C:\WINDOWS\SYSTEM32\ntdll.dll
Report Id: 735b6b56-91dc-4669-8393-1fc6434defb7
Faulting package full name: 
Faulting package-relative application ID: 
Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Application Error" Guid="{a0e9b465-b939-57d7-b27d-95d8e925ff57}" />
    <EventID>1000</EventID>
    <Version>0</Version>
    <Level>2</Level>
    <Task>100</Task>
    <Opcode>0</Opcode>
    <Keywords>0x8000000000000000</Keywords>
    <TimeCreated SystemTime="2025-02-27T21:35:31.7795452Z" />
    <EventRecordID>1236</EventRecordID>
    <Correlation />
    <Execution ProcessID="26008" ThreadID="8204" />
    <Channel>Application</Channel>
    <Computer>JDMILS02</Computer>
    <Security UserID="S-1-5-21-1503500844-3549604201-1630939383-1001" />
  </System>
  <EventData>
    <Data Name="AppName">freecad.exe</Data>
    <Data Name="AppVersion">1.1.0.0</Data>
    <Data Name="AppTimeStamp">67b8deca</Data>
    <Data Name="ModuleName">ntdll.dll</Data>
    <Data Name="ModuleVersion">10.0.26100.3037</Data>
    <Data Name="ModuleTimeStamp">95e6c489</Data>
    <Data Name="ExceptionCode">c0000005</Data>
    <Data Name="FaultingOffset">0000000000163ea7</Data>
    <Data Name="ProcessId">0x3dd4</Data>
    <Data Name="ProcessCreationTime">0x1db895e769d7e66</Data>
    <Data Name="AppPath">C:\Program Files\FreeCAD_weekly-builds-40317-conda-Windows-x86_64-py311\bin\freecad.exe</Data>
    <Data Name="ModulePath">C:\WINDOWS\SYSTEM32\ntdll.dll</Data>
    <Data Name="IntegratorReportId">735b6b56-91dc-4669-8393-1fc6434defb7</Data>
    <Data Name="PackageFullName">
    </Data>
    <Data Name="PackageRelativeAppId">
    </Data>
  </EventData>
</Event>

Log Name:      Application
Source:        Windows Error Reporting
Date:          28/02/2025 8:35:35 AM
Event ID:      1001
Task Category: None
Level:         Information
Keywords:      
User:          JDMILS02\julian
Computer:      JDMILS02
Description:
Fault bucket 1695013735773624175, type 4
Event Name: APPCRASH
Response: Not available
Cab Id: 0

Problem signature:
P1: freecad.exe
P2: 1.1.0.0
P3: 67b8deca
P4: ntdll.dll
P5: 10.0.26100.3037
P6: 95e6c489
P7: c0000005
P8: 0000000000163ea7
P9: 
P10: 

Attached files:
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.fa7ec654-2e34-4b18-8db2-ecc1940285c8.tmp.dmp
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.f0c5f55b-59d1-46e5-a8b8-4c405836d859.tmp.WERInternalMetadata.xml
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.d2fbae8f-593d-4fd9-9610-5e34f7f33c77.tmp.csv
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.3f726dea-da07-45c8-9007-fae7102046ed.tmp.txt
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.7fceb2bf-7b45-49d5-8cc4-ad45b139565f.tmp.xml

These files may be available here:
\\?\C:\ProgramData\Microsoft\Windows\WER\ReportArchive\AppCrash_freecad.exe_54717cffbc85a2e28185f2bb3a92b0351b2b2313_f4f6cb61_ae3767a9-6100-4dc5-a54f-d82642718526

Analysis symbol: 
Rechecking for solution: 0
Report Id: 735b6b56-91dc-4669-8393-1fc6434defb7
Report Status: 268435456
Hashed bucket: 446776d7a1ae1c0e8785e6031a31d76f
Cab Guid: 0
Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Windows Error Reporting" Guid="{0ead09bd-2157-539a-8d6d-c87f95b64d70}" />
    <EventID>1001</EventID>
    <Version>0</Version>
    <Level>4</Level>
    <Task>0</Task>
    <Opcode>0</Opcode>
    <Keywords>0x8000000000000000</Keywords>
    <TimeCreated SystemTime="2025-02-27T21:35:35.8942598Z" />
    <EventRecordID>1237</EventRecordID>
    <Correlation />
    <Execution ProcessID="26008" ThreadID="8204" />
    <Channel>Application</Channel>
    <Computer>JDMILS02</Computer>
    <Security UserID="S-1-5-21-1503500844-3549604201-1630939383-1001" />
  </System>
  <EventData>
    <Data Name="Bucket">1695013735773624175</Data>
    <Data Name="BucketType">4</Data>
    <Data Name="EventName">APPCRASH</Data>
    <Data Name="Response">Not available</Data>
    <Data Name="CabId">0</Data>
    <Data Name="P1">freecad.exe</Data>
    <Data Name="P2">1.1.0.0</Data>
    <Data Name="P3">67b8deca</Data>
    <Data Name="P4">ntdll.dll</Data>
    <Data Name="P5">10.0.26100.3037</Data>
    <Data Name="P6">95e6c489</Data>
    <Data Name="P7">c0000005</Data>
    <Data Name="P8">0000000000163ea7</Data>
    <Data Name="P9">
    </Data>
    <Data Name="P10">
    </Data>
    <Data Name="AttachedFiles">
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.fa7ec654-2e34-4b18-8db2-ecc1940285c8.tmp.dmp
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.f0c5f55b-59d1-46e5-a8b8-4c405836d859.tmp.WERInternalMetadata.xml
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.d2fbae8f-593d-4fd9-9610-5e34f7f33c77.tmp.csv
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.3f726dea-da07-45c8-9007-fae7102046ed.tmp.txt
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER.7fceb2bf-7b45-49d5-8cc4-ad45b139565f.tmp.xml</Data>
    <Data Name="StorePath">\\?\C:\ProgramData\Microsoft\Windows\WER\ReportArchive\AppCrash_freecad.exe_54717cffbc85a2e28185f2bb3a92b0351b2b2313_f4f6cb61_ae3767a9-6100-4dc5-a54f-d82642718526</Data>
    <Data Name="AnalysisSymbol">
    </Data>
    <Data Name="Rechecking">0</Data>
    <Data Name="ReportId">735b6b56-91dc-4669-8393-1fc6434defb7</Data>
    <Data Name="ReportStatus">268435456</Data>
    <Data Name="HashedBucket">446776d7a1ae1c0e8785e6031a31d76f</Data>
    <Data Name="CabGuid">0</Data>
  </EventData>
</Event>`

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40317 (Git) Conda
Build type: Release
Branch: main
Hash: b67ab979afdba808fc03196faf9c5f8830df471f
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/Australia (en_AU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Logical/physical DPI: 96/106.57
Installed mods: 
  * A2plus 0.4.68
  * Assembly4 0.50.15
  * Curves 0.6.58
  * fasteners 0.5.36
  * Gridfinity 0.8.0
  * lattice2 1.0.0
```

### Subproject(s) affected?

None

### Anything else?

I tried the same in FC 39980 and it does not crash, however the model fails with Report Viewer showing:
`08:39:02  Cut: Boolean operation failed
Sketch_On_Surface is not a solid`

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

jffmichi, 2025-02-28:
Not entirely sure what file you are referring to. I could not reproduce the crash in `Cut an angle out of a curved block_JDMILS_tarman.FCStd` in case that's the one.

However, here is a file that triggers the issue for me:
[boolean_cut_crash.zip](https://github.com/user-attachments/files/19019306/boolean_cut_crash.zip)

I hope that PR #19911 also solves the issue for you...

mnesarco, 2025-03-04:
I get a segfault on recompute and i suspect it is related with this issue, here is part of the stack trace:

```bash
<TopoShape> TopoShapeExpansion.cpp(1412): Not all input shapes are mappable
<TopoShape> TopoShapeExpansion.cpp(1412): Not all input shapes are mappable

Thread 1 "FreeCAD" received signal SIGSEGV, Segmentation fault.
0x00007ffff12a6f67 in _int_malloc () from /nix/store/wn7v2vhyyyi6clcyn0s9ixvl7d4d87ic-glibc-2.40-36/lib/libc.so.6
(gdb) bt
#0  0x00007ffff12a6f67 in _int_malloc () from /nix/store/wn7v2vhyyyi6clcyn0s9ixvl7d4d87ic-glibc-2.40-36/lib/libc.so.6
#1  0x00007ffff12a95d8 in calloc () from /nix/store/wn7v2vhyyyi6clcyn0s9ixvl7d4d87ic-glibc-2.40-36/lib/libc.so.6
#2  0x00007fffbfeb3bb0 in Standard::Allocate(unsigned long) () from /nix/store/x6jaqqm5c0mayqyn65m0f25rb4paaxza-opencascade-occt-7.8.1/lib/libTKernel.so.7.8
#3  0x00007fffbf3f65cf in NCollection_ListNode::operator new (theAllocator=..., theSize=32) at /nix/store/x6jaqqm5c0mayqyn65m0f25rb4paaxza-opencascade-occt-7.8.1/include/opencascade/NCollection_ListNode.hxx:30
#4  NCollection_List<TopoDS_Shape>::appendList (pCur=0x1372ff00, this=0x7fffff7ff100) at /nix/store/x6jaqqm5c0mayqyn65m0f25rb4paaxza-opencascade-occt-7.8.1/include/opencascade/NCollection_List.hxx:380
#5  NCollection_List<TopoDS_Shape>::NCollection_List (this=0x7fffff7ff100, theOther=...) at /nix/store/x6jaqqm5c0mayqyn65m0f25rb4paaxza-opencascade-occt-7.8.1/include/opencascade/NCollection_List.hxx:71
#6  0x00007fffbf3f5e1a in FCBRepAlgoAPI_BooleanOperation::Build (this=0x14a934f0) at /FreeCAD/FreeCAD_nix/FreeCAD/src/Mod/Part/App/FCBRepAlgoAPI_BooleanOperation.cpp:103
#7  0x00007fffbf3f61f1 in FCBRepAlgoAPI_BooleanOperation::Build (this=0x14a934f0) at /FreeCAD/FreeCAD_nix/FreeCAD/src/Mod/Part/App/FCBRepAlgoAPI_BooleanOperation.cpp:123
#8  0x00007fffbf3f61f1 in FCBRepAlgoAPI_BooleanOperation::Build (this=0x14a934f0) at /FreeCAD/FreeCAD_nix/FreeCAD/src/Mod/Part/App/FCBRepAlgoAPI_BooleanOperation.cpp:123
#9  0x00007fffbf3f61f1 in FCBRepAlgoAPI_BooleanOperation::Build (this=0x14a934f0) at /FreeCAD/FreeCAD_nix/FreeCAD/src/Mod/Part/App/FCBRepAlgoAPI_BooleanOperation.cpp:123
#10 0x00007fffbf3f61f1 in FCBRepAlgoAPI_BooleanOperation::Build (this=0x14a934f0) at /FreeCAD/FreeCAD_nix/FreeCAD/src/Mod/Part/App/FCBRepAlgoAPI_BooleanOperation.cpp:123
#11 0x00007fffbf3f61f1 in FCBRepAlgoAPI_BooleanOperation::Build (this=0x14a934f0) at /FreeCAD/FreeCAD_nix/FreeCAD/src/Mod/Part/App/FCBRepAlgoAPI_BooleanOperation.cpp:123
#12 0x00007fffbf3f61f1 in FCBRepAlgoAPI_BooleanOperation::Build (this=0x14a934f0) at /FreeCAD/FreeCAD_nix/FreeCAD/src/Mod/Part/App/FCBRepAlgoAPI_BooleanOperation.cpp:123
#13 0x00007fffbf3f61f1 in FCBRepAlgoAPI_BooleanOperation::Build (this=0x14a934f0) at /FreeCAD/FreeCAD_nix/FreeCAD/src/Mod/Part/App/FCBRepAlgoAPI_BooleanOperation.cpp:123
#14 0x00007fffbf3f61f1 in FCBRepAlgoAPI_BooleanOperation::Build (this=0x14a934f0) at /FreeCAD/FreeCAD_nix/FreeCAD/src/Mod/Part/App/FCBRepAlgoAPI_BooleanOperation.cpp:123
#15 0x00007fffbf3f61f1 in FCBRepAlgoAPI_BooleanOperation::Build (this=0x14a934f0) at /FreeCAD/FreeCAD_nix/FreeCAD/src/Mod/Part/App/FCBRepAlgoAPI_BooleanOperation.cpp:123
#16 0x00007fffbf3f61f1 in FCBRepAlgoAPI_BooleanOperation::Build (this=0x14a934f0) at /FreeCAD/FreeCAD_nix/FreeCAD/src/Mod/Part/App/FCBRepAlgoAPI_BooleanOperation.cpp:123
#17 0x00007fffbf3f61f1 in FCBRepAlgoAPI_BooleanOperation::Build (this=0x14a934f0) at /FreeCAD/FreeCAD_nix/FreeCAD/src/Mod/Part/App/FCBRepAlgoAPI_BooleanOperation.cpp:123
#18 0x00007fffbf3f61f1 in FCBRepAlgoAPI_BooleanOperation::Build (this=0x14a934f0) at /FreeCAD/FreeCAD_nix/FreeCAD/src/Mod/Part/App/FCBRepAlgoAPI_BooleanOperation.cpp:123
#19 0x00007fffbf3f61f1 in FCBRepAlgoAPI_BooleanOperation::Build (this=0x14a934f0) at /FreeCAD/FreeCAD_nix/FreeCAD/src/Mod/Part/App/FCBRepAlgoAPI_BooleanOperation.cpp:123
#20 0x00007fffbf3f61f1 in FCBRepAlgoAPI_BooleanOperation::Build (this=0x14a934f0) at /FreeCAD/FreeCAD_nix/FreeCAD/src/Mod/Part/App/FCBRepAlgoAPI_BooleanOperation.cpp:123
...

```

The rest is the same 1000+ apparently an infinite recursion...

JDMils, 2025-03-04:
FYI, I'm using my original file as posted in the forum topic:
https://forum.freecad.org/download/file.php?id=294273
Name: [Cut an angle out of a curved block_JDMILS.FCStd](https://forum.freecad.org/download/file.php?id=294273)

I tested the process on FC 39980 and it did not crash. I just get the expected error:
```
16:36:09  Cut: Boolean operation failed
Sketch_On_Surface is not a solid
```
This is expected for this type of operation.

However, when I run the same process on FC 40317, it crashes to the desktop. I've included a GIF of my screen when the issue happens.

[image]

In summary, to me this indicates that some sort of instability has been introduced from FC 39980 to FC 40317. And IMO, I've seen other instabilities in the app which never happened before so something is not right.

closed 2025-03-11 by chennes