FreeCAD/FreeCAD #19914: Message starting FreeCAD

Beto-Osorio, 2025-02-28:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The following message appears every time I start FreeCAD:

08:53:44  DirectWrite: CreateFontFaceFromHDC() failed (Indicates an error in an input file such as a font file.) for QFontDef(Family="8514oem", pointsize=10, pixelsize=20, styleHint=5, weight=50, stretch=100, hintingPreference=0) LOGFONT("8514oem", lfWidth=0, lfHeight=-20) dpi=120

### Full version info

```shell
[code]
OS: Windows 11 build 26100
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * DesignSPHysics 0.7.0(15-09-2023)
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

DualPhysics Macro doesn't work properly

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-28:
You are using an outdated version. Please use 1.0 or test in the weekly 1.1 version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
Please try restart in safe mode too.

Syres916, 2025-02-28:
All you need to do is Edit > Preferences > Python > Editor tab > Family change to a suitable font and click OK

[image]

luzpaz, 2025-06-02:
@julian1 does #21710 address this ?