FreeCAD/FreeCAD #19915: Drop-down menu items are not displayed on wayland.

kberkut, 2025-02-28:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Drop-down menu items are not displayed on wayland.

https://github.com/user-attachments/assets/2bd8d150-045a-4084-b6a7-1fbb52df682b

## System

OS: Arch Linux x86_64 
Resolution: 1920x1080 
DE: GNOME 47.4 
CPU: AMD Ryzen 5 3500U with Radeon Vega Mobile Gfx (8) @ 2.100GHz 
GPU: AMD ATI Radeon Vega Series / Radeon Vega Mobile Series 
Memory: 5676MiB / 17921MiB 

### Full version info

```shell
OS: Arch Linux (GNOME/gnome/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.13.2, Qt 6.8.2, Coin 4.0.3, Vtk 9.4.1, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-02-28:
Related: https://github.com/FreeCAD/FreeCAD/issues/19352

luzpaz, 2025-05-15:
Is this still an issue ?
I can't reproduce on: 

```
OS: Manjaro Linux (KDE/plasmawayland/wayland)
Architecture: x86_64
Version: 1.1.0dev.41694 (Git) AppImage
Build date: 2025/05/13 12:46:54
Build type: Release
Branch: main
Hash: 5e624a4775c993a3d940f37876840c65ac0cc60c
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/149.623/1
Installed mods: 
  * fcVM
  * fcVM.backup1747322268.288692 (Disabled)
  * FreeCAD-Ribbon 1.9.1
  * SearchBar 1.4.2
  * Silk 0.1.8
  * Telemetry 1.0.1
```

kberkut, 2025-05-16:
In version 1.0.1.39285 (Git) Conda AppImage:
OS: Arch Linux (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion

Work perfectly!