FreeCAD/FreeCAD #19919: VarSet: VarSet linked from another file does not autocomplete.

leoheck, 2025-03-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

It could autocomplete since the properties are visible from the linked file.

I am trying to centralize VarSets.
So I created a file to put them.
I want to import (link the VarSet) on other files so the variables are widely accessible through a multifile project.

### Full version info

```shell
1.1-dev 40444
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

furgo16, 2025-03-01:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/12252. See https://github.com/FreeCAD/FreeCAD/issues/12252#issuecomment-2660780910

leoheck, 2025-03-01:
I used to model everything on the same file, and now I am splitting into multiple files.
Ahaha, we are walking the same way but each one following a different track.
My files were getting so complex that every change would recalculate many things and break many places that I started to thing it would be better to split everything in files. But Freecad is not quite prepared for a project-like approach.

closed 2025-03-02 by maxwxyz