FreeCAD/FreeCAD #19920: Sketcher - Idea - Constraints could have an option to display the name under the value.

leoheck, 2025-03-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Constraints could have an option to display the name under the value.
The same could work for equations.

From this:
[image]

to this:
[image]

This could be an interesting way to display where the properties of VarSets are located.
You can draw them on a Sketch, and then enable visibility of the equation.
then, for instance, it could show:

[image]

### Full version info

```shell
1.1-dev 40444
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

kaiwas, 2025-03-01:
overall not bad.
only the expressions can be very very very long. sometimes they don't fit the entire width of the screen.
some adaptive option is needed

pierreporte, 2025-03-01:
There is already a proposal for this. See https://github.com/FreeCAD/FreeCAD/issues/12036#issuecomment-1946236285.

Showing the expression is not a good idea. Your examples only work because they are very simple.

furgo16, 2025-03-01:
> Showing the expression is not a good idea. Your examples only work because they are very simple.

I'd agree. The expression itself could quickly clutter the view. What about just showing the f(x) icon to give the user a visual cue that the constraint is driven by an expression?

leoheck, 2025-03-01:
> overall not bad. only the expressions can be very very very long. sometimes they don't fit the entire width of the screen. some adaptive option is needed.
> 
> maybe it could be a hover tooltip

This is the user decision. 
They can use a simple variable and display the property name. And they don't have to show the property name of the expression if it is big. Also, if the expression is big it is better to have a new variable with the expression. The complexity is defined by the user. I am just proposing to show the name and the value, independently if the user wants that.

leoheck, 2025-03-01:
> Textual seems cleaner though.

I think I prefer my suggestion, since cleaner means it is not quite an improvement. And one line means cluttered. The user will use this whenever is convenient they doesn't need to show the property and the value everytime. Also, I would use this on special sketches where I am defining and displaying where each variable is being used.

pierreporte, 2025-03-01:
> This is the user decision. They can use a simple variable and display the property name. And they don't have to show the property name of the expression if it is big. Also, if the expression is big [...]

If the user needs to put workaround in place because the display is sub-optimal, then it’s a bad idea from the start.

> I think I prefer my suggestion, since cleaner means it is not quite an improvement. And one line means cluttered. The user will use this whenever is convenient they doesn't need to show the property [...]

I too believe that your UI suggestion is less clean, less clear, and overall too distracting. The proposal shown in #12036 is basically writing a simple mathematical relation as someone would do on paper. It is also the same thin.

On top of that, your proposal introduces new color while @FreeCAD/design-working-group is working towards reducing the number of colors in sketcher. The proposal mentioned in #12036 basically removes all colors but one (the full sketcher proposal reomves the separate color for dimensions and constrains to use the fully-defined geometry color as in Catia).

Though it would be a good idea to have a parameter to hide all dimension names.

leoheck, 2025-03-01:
> If the user needs to put workaround in place because the display is sub-optimal, then it’s a bad idea from the start.

I don't consider a workaround.
I consider that, if the expression is too big, maybe it is time to set a new variable with that thing.
It is more like a design style as in code style, right?

> I too believe that your UI suggestion is less clean, less clear, and overall too distracting.

Ok, no problem. Having the property names would be a good start. We may be thinking of it in different scenarios.
As I could understand you are describing something that is used by default. I am describing something the user enables on the go when needed. My idea is to have such a master sketch that is self-documented so you can define and keep in one place, the variable's names close to what they are defining. I don't think I would use it on every Sketch. Since both proposals clutter the view.

> On top of that, your proposal introduces a new color ...

Ah proposal, is the 2 options that the user can enable/disable independently.
The style can be discussed, of course, especially of the colors.

> Though it would be a good idea to have a parameter to hide all dimension names.

Not only the names (properties) but the value (fx) too, right?

maxwxyz, 2025-03-01:
I'm closing this in favor of #12036 maybe discussion can continue there

closed 2025-03-01 by maxwxyz