FreeCAD/FreeCAD #19931: incompatibility of versions

Denis461, 2025-03-02:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Good afternoon. When recalculating the model, an error appears stating that recalculation is not possible. The model was recalculated from version 0.19 to 0.21. But it doesn't want to switch to 1.0. The rounding error (Fillet).

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * fasteners 0.5.33
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

szdowk, 2025-03-07:
I downloaded a second of attached files. After initial opening, it crash Freecad while recomputing...

OS: Slackware 15.0 x86_64 (KDE/xcb)
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/81.5973
Installed mods: 
  * sheetmetal 0.6.13
  * Help 1.0.3
  * freecad.gears 1.3.0
  * Curves 0.6.53

Denis461, 2025-03-08:
It was drawn at 0.19. It opens and edits without problems at 0.21. But he doesn't want to be in the new one. Do you know how to fix it?

xtemp09, 2025-03-08:
Here is the backtrace.

[backtrace.txt](https://github.com/user-attachments/files/19142379/backtrace.txt)

```ruby
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40607 (Git)
Build date: 2025/03/08 01:10:58
Build type: Debug
Branch: master
Hash: 50e80f4a03ac9eda45e2c3f80fde26d9121ad15e
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Logical DPI/Physical DPI/Pixel Ratio: 96.0584/96.0584/1
```

Denis461, 2025-03-08:
I don't really understand it. And how to use it?

xtemp09, 2025-03-08:
> I don't really understand it. And how to use it?

It is used by those who understand it. =)

chennes, 2025-06-30:
@pieterhijma can you see if this is an OCCT issue?

closed 2026-08-15 by maxwxyz