FreeCAD/FreeCAD #19935: Carbon copying a non-parallel sketch has incorrect help text on macOS

ajh0912, 2025-03-02:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In the sketcher using the carbon copy tool, if you hover over a sketch from a non-parallel plane, the help text in the bottom left says `The selected sketch is not parallel to this sketch. Hold Ctrl+Alt to allow non-parallel sketches.`

On macOS this shortcut is actually CMD+Option.

### Full version info

```shell
OS: macOS 15.0
Architecture: arm64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

Sketcher

### Anything else?

Tutorial being followed: https://youtu.be/sgzKC5moK_M?t=1046

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-09:
@FreeCAD/design-working-group any idea what could be a better wording here?

cteubank, 2025-07-20:
I hit the same problem. Can confirm as per the original poster that CMD+Option is the keyboard combo that enables non-parallel sketches in the Carbon Copy tool on macOS.

luzpaz, 2025-07-20:
https://github.com/FreeCAD/FreeCAD/blob/6974c83f9a7a30c13d750543797a2e7520d251d0/src/Mod/Sketcher/Gui/DrawSketchHandlerCarbonCopy.h#L97-L108

There are a few tooltips that can be addressed in the same file. 

Maybe putting (CMD+Option) in parenthesis ?

luzpaz, 2026-09-29:
> > We're going to defer this to 1.2 to try to figure out what an actual good path forward is.
> 
> Per chennes in [#23960 (comment)](https://github.com/FreeCAD/FreeCAD/pull/23960#issuecomment-3347722438)
> 

@maxwxyz this fell off the roadmap.

anshika170-cmyk, 2026-10-05:
Hi I want to work on this

BackendArchitectX, 2026-10-14:
I'm working on a fix for this issue.

My source-level investigation identified four Carbon Copy rejection messages with hardcoded `Ctrl` / `Ctrl+Alt` instructions, which do not reflect the corresponding macOS modifiers. I also identified a translation-context mismatch: the messages are defined under `SketcherGui::CarbonCopySelection`, while both hover and click rejection paths use `QObject::tr()`.

I'm pursuing a placeholder-based solution consistent with the feedback on #23960. The objective is to maintain a single translatable sentence per rejection reason, display platform-appropriate modifier labels, and preserve existing Carbon Copy selection behavior.

The planned regression coverage includes:

- All eight Carbon Copy rejection reasons.
- Correct modifier labels across supported platforms.
- Consistent translation and formatting for hover and click.
- Translation fallback and placeholder handling.
- Compatibility with existing selection gates.

The source-level audit has passed. Implementation, compilation, and runtime validation, including macOS verification, are still pending.

I'll continue with the implementation and tests and submit a focused PR for review.