FreeCAD/FreeCAD #19936: BIM: Arch_Space not working for concave spaces

maxwxyz, 2025-03-02:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

From https://github.com/FreeCAD/FreeCAD/issues/18917

The Arc_Space commands fails/creates a wrong geometry where 2 walls meet at an angle like this, even when I join the walls):
[image]

Even like this, where I have not intersecting boundaries:
[image]

 @paullee0 mentioned on the PR:

> I think it is the current algorithm limitation like 'concave' case (faces with angle > 180 deg) Yorik mentioned.
> 
> That's why I made some improvement with a few videos below (oh, 3 years ago) how Spaces could be made automatically without such problems :
> 
> https://www.youtube.com/watch?v=0bnQZ473R7A
> https://www.youtube.com/watch?v=Ob78XVOnmXQ
> https://www.youtube.com/watch?v=1XR1I9WG75Y
> https://www.youtube.com/watch?v=6E_rgq762Q8

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40444 (Git) Conda
Build type: Release
Branch: main
Hash: 089605bb4db98668661bf56db3f5f5bb4f7f0d87
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/108.888
Installed mods: 
  * Cables 0.1.3
  * CfdOF 1.29.8
  * CurvedShapes 1.0.13
  * Curves 0.6.58
  * dodo 1.0.1
  * fasteners 0.5.37
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.1.2
  * parts_library
  * Rocket 4.1.1
  * sheetmetal 0.7.20
  * Silk 0.1.5
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

paullee0, 2025-03-02:
Indeed this is tricky, my approaches are based on Sketch / ArchSketch.  Not much time to refine it and expose it in the add-on thoug.

Hope there are ideas which work one workflow without Sketch base, anyone?

maxwxyz, 2025-03-02:
Maybe @yorikvanhavre has an idea based on the new walls without base objects?

Also in such a simple cubic case, the Arch_Space command also fails and creates something different:

[image]

Roy-043, 2025-03-02:
Duplicate of #6091.

closed 2025-03-02 by maxwxyz