FreeCAD/FreeCAD #19938: App: different version info output

benj5378, 2025-03-02:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Using GUI workflow Help -> About -> Copy to clipboard outputs

```
OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.40422 (Git)
Build type: Debug
Branch: (HEAD detached at 56f0f64015)
Hash: 56f0f640153926552effff9a3e931c8ce9a49e11
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/159.34
Installed mods: 
  * OpenTheme 2024.9.1
  * Behave-Dark-Colors 0.1.1
```

which is different from using `freecad -v --verbose` that gives

```
FreeCAD 1.1.0 Revision: 40422 (Git)

Library versions:
boost    1_83
Coin3D   4.0.2
Eigen3   3.4.0
OCC      7.6.3
Qt       5.15.13
Python   3.12.3
PySide   5.15.13
shiboken 5.15.13
SMESH    7.7.1.0
VTK      9.1.0
xerces-c 3.2.4
```

### Full version info

```shell
0c65673a6fd2421be8fbe613116077120adea068
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

paulywill, 2025-03-02:
You want the information presented to be the same format?

benj5378, 2025-03-02:
@paulywill it should just be identical

furgo16, 2025-03-02:
I'd agree with the issue description. I cannot see a motivation for the output to be different. I guess having identical output would simplify the code too.

To me, that would be ideal for bug reporting as well. Every time the reporter needs to add the FreeCAD About info they need to launch the full UI, wait until it's loaded and then navigate to the right location. If `freecad -v --verbose` produced the same output, it could output and copied in 1-2 seconds instead.

benj5378, 2025-03-02:
Idk why it isn't just verbose by default

luzpaz, 2025-03-13:
Anyone want to spearhead this ?

xelathan, 2025-03-25:
@luzpaz I can pick it up if no one has already

benj5378, 2025-03-25:
@xelathan please do

xelathan, 2025-03-26:
@benj5378 moving most of the info from the About Dialog to the Application.cpp to display through the CLI went fine but I ran into some issues trying to display the DPI info because it uses QApplication and the App module seems to operate in a non QtGui context. do you know anyway to retrieve this information?

benj5378, 2025-03-26:
@xelathan I'd keep that info in Gui then

xelathan, 2025-03-26:
Here is the corresponding PR: #20487

PR #20487 (2025-04-16): App: Running FreeCAD in verbose mode information to reflect Gui -> Help -> About Dialog info
Fixes: #19938 

This PR unifies the output displayed on Gui Help -> About -> Copy to clipboard outputs to running `freecad` with `--verbose` flag. Running `freecad` with `--verbose` flag should now output the same information as the Gui About Dialog copied information.

Below is an example of the changes:

`FreeCAD --verbose --version`

[image]

Gui Help -> About -> Copy to clipboard
```
OS: Pop!_OS 22.04 LTS (Unity/pop/xcb)
Architecture: x86_64
Version: 1.1.0dev.41029 (Git)
Build date: 2025/03/28 03:2 [... cut]
closed 2025-04-16 by hyarion