FreeCAD/FreeCAD #19952: CAM: closes unexpectedly when applying any operation

picli3, 2025-03-02:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I'm trying the new release in development and to generate Gcodes files for my CNC, but when applying some operation it closes unexpectedly.

I was able to capture the error.

### Full version info

```shell
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.40408 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: b07caa732e63064b8851ecdda184292962d95c15
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: Spanish/Peru (es_PE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/80.9812
Installed mods: 
  * sheetmetal 0.7.20
  * ArchTextures
  * FreeCAD-themes 2025.1.7
  * fasteners 0.5.37
  * Curves 0.6.58
  * kicadStepUpMod 11.5.2
```

### Subproject(s) affected?

CAM

### Anything else?

[Cajita_front.zip](https://github.com/user-attachments/files/19045233/Cajita_front.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

tarman3, 2025-03-03:
Confirm
In build 40408 CAM workbench working terrible unstable

maxwxyz, 2025-03-03:
@jbaehr @sliptonic FYI

tarman3, 2025-03-04:
Looks like my problem as solved in FreeCAD 1.1.0dev.40504

<details>
  <summary>FreeCAD 1.1.0dev.40504 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/118.293
Installed mods: CurvedShapes 1.0.13; lattice2 1.0.0; Silk 0.1.5; freecad.gears 1.3.0; btl 0.9.9 (Disabled); fasteners 0.5.37; MeshRemodel 1.10.36; Curves 0.6.58; sheetmetal 0.7.11; Defeaturing 1.2.3

```

</details>

maxwxyz, 2025-03-04:
@picli3 can you please check on the latest weekly and report back? If there is still an issue, I'll reopen it.

closed 2025-03-04 by maxwxyz