FreeCAD/FreeCAD #19954: CAM:  Many refactored* postprocessor configuration choices don't have command line options

LarryWoestman, 2025-03-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The refactored* postprocessor shared code base has many configuration choices that don't have command line options.  This "hides" some of the capabilities of the code base and forces people to write code to use many configuration choices.

My goal is that all configuration choices (that make sense) should have command line options with corresponding tooltip info and tests.

I will be filing several PR's to add the command line options, each PR containing a few so as to not get too big.

Please assign this issue to me.  Thanks!

### Full version info

```shell
All FreeCAD versions up to this point.
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

LarryWoestman, 2025-03-03:
PR #19955 partially fixes this issue.

luzpaz, 2025-03-31:
@LarryWoestman would you mind posting what is left to do since #19955 was merged ?

LarryWoestman, 2025-03-31:
There are probably another 20 or so configuration choices that don't have command line options at the moment.  I expect to deal with 3 or 4 of them in each PR, which means somewhere around 4 to 6 more PR's (to keep them relatively small).

I plan to create another PR this week, if things go as planned.

LarryWoestman, 2025-04-16:
#20833 add 3 more command line options.

LarryWoestman, 2025-05-25:
#21563 adds 3 more command line options.

LarryWoestman, 2025-06-08:
#21881 adds 3 more command line options.