FreeCAD/FreeCAD #19956: Running on Linux Fedora 36 and generation.py failing to generate headers / XML files for cmake buildfiles generation...

DaemonDave, 2025-03-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

```
--- /home/dave/src/FreeCAD/src/Base/BoundBoxPy.h+++ /home/dave/src/FreeCAD/src/Base/BoundBoxPy_.h@@ -1,15 +1,15 @@ 
 // This file is generated by src/Tools/generateTemplates/templateClassPyExport.py out of the XML file
 // Every change you make here gets lost in the next full rebuild!
-#ifndef BASE_BOUNDBOXPY_H
-#define BASE_BOUNDBOXPY_H
-
-#include <Base/PyObjectBase.h>
-#include <Base/BoundBox.h>
+#ifndef FREECAD_BOUNDBOXPY_H
+#define FREECAD_BOUNDBOXPY_H
+
+#include <Mod/FreeCAD/PyObjectBase.h>
+#include <Mod/FreeCAD/BoundBox.h>
 #include <string>
 
 
-namespace Base
+namespace FreeCAD
 {
 
 //===========================================================================
@@ -18,7 +18,7 @@ 
 /** The python export class for BoundBox
  */
-class BaseExport BoundBoxPy : public Base::PyObjectBase
+class FreeCADExport BoundBoxPy : public FreeCAD::PyObjectBase
 {
 protected:
     ~BoundBoxPy() override;
@@ -209,8 +209,8 @@ 
 };
 
-}  //namespace Base
-
-#endif  // BASE_BOUNDBOXPY_H
-
-
+}  //namespace FreeCAD
+
+#endif  // FREECAD_BOUNDBOXPY_H
+
+

CMake Error at cMake/FreeCadMacros.cmake:159 (execute_process):
  execute_process failed command indexes:

    1: "Child return code: 1"

Call Stack (most recent call first):
  src/Base/CMakeLists.txt:86 (generate_from_py)

-- Configuring incomplete, errors occurred!
```

Running on Linux Fedora 36 and generation.py failing to generate headers / XML files for cmake buildfiles generation...

How to solve this?

### Full version info

```shell
FreeCad-main git cloned today...
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

Ran cmake and git submodule init before failure.

What's interesting is that when I ran cmake on the downloaded .zip first but it didn't fail at generate.py ...

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

furgo16, 2025-03-03:
Are there any relevant errors in the log before what you pasted?

This seems similar to this failure on yesterday's FreeCAD-snap daily build: https://github.com/FreeCAD/FreeCAD-snap/issues/178

DaemonDave, 2025-03-03:
Thanks for the quick response:

Here's the cmake configure output:

-- Found Pivy 0.6.8
-- Checking Pivy Coin3D version by importing it in a Python program...
-- Found Pivy Coin3D SIM Coin 4.0.0
-- Could NOT find Spnav (missing: SPNAV_LIBRARY SPNAV_INCLUDE_DIR) 
-- Found pip-installed Shiboken2 in /usr/lib64/python3.10/site-packages/shiboken2
-- Found pip-installed PySide2 in /usr/lib64/python3.10/site-packages/PySide2
CMake Warning at cMake/FreeCAD_Helpers/SetupShibokenAndPyside.cmake:128 (message):
  Shiboken2 include files not found, FREECAD_USE_SHIBOKEN automatically set
  to OFF
Call Stack (most recent call first):
  CMakeLists.txt:111 (SetupShibokenAndPyside)

CMake Warning at cMake/FreeCAD_Helpers/SetupShibokenAndPyside.cmake:154 (message):
  PySide2 include files not found, FREECAD_USE_PYSIDE automatically set to
  OFF
Call Stack (most recent call first):
  CMakeLists.txt:111 (SetupShibokenAndPyside)

-- PySide 5.15.7 Python module found at /usr/lib64/python3.10/site-packages/PySide2.

-- Found Matplotlib: /usr/lib64/python3.10/site-packages/matplotlib (found version "3.5.3")
-- Found Boost: /usr/include (found suitable version "1.76.0", minimum required is "1.74") found components: program_options regex system thread date_time chrono atomic 
-- Platform is 64-bit, set -D_OCC64
-- Performing Test _flag_found
-- Performing Test _flag_found - Success
-- Performing Test _flag_found
-- Performing Test _flag_found - Success
-- Performing Test _flag_found
-- Performing Test _flag_found - Success
-- Performing Test _flag_found
-- Performing Test _flag_found - Success
-- [E57] Revison ID: E57Format-2.2.1-x86_64-linux-gcc12
-- [E57] Building static library
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY - Success
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY - Success
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
-- [OndselSolver] Building shared library
-- /home/dave/src/FreeCAD/src/Base/BoundBox.cpp
Traceback (most recent call last):
  File "/home/dave/src/FreeCAD/src/Tools/bindings/generate.py", line 112, in <module>
    main()
  File "/home/dave/src/FreeCAD/src/Tools/bindings/generate.py", line 108, in main
    generate(filename, outputPath)
  File "/home/dave/src/FreeCAD/src/Tools/bindings/generate.py", line 63, in generate
    Export.Compare()
  File "/home/dave/src/FreeCAD/src/Tools/bindings/templates/templateClassPyExport.py", line 106, in Compare
    compareFiles(outputHeaderXml, outputHeaderPy)
  File "/home/dave/src/FreeCAD/src/Tools/bindings/templates/templateClassPyExport.py", line 34, in compareFiles
    raise ValueError(error)
ValueError: Files are not equal.

Diff:

--- /home/dave/src/FreeCAD/src/Base/BoundBoxPy.h+++ /home/dave/src/FreeCAD/src/Base/BoundBoxPy_.h@@ -1,15 +1,15 @@ 
 // This file is generated by src/Tools/generateTemplates/templateClassPyExport.py out of the XML file
 // Every change you make here gets lost in the next full rebuild!
-#ifndef BASE_BOUNDBOXPY_H
-#define BASE_BOUNDBOXPY_H
-
-#include <Base/PyObjectBase.h>
-#include <Base/BoundBox.h>
+#ifndef FREECAD_BOUNDBOXPY_H
+#define FREECAD_BOUNDBOXPY_H
+
+#include <Mod/FreeCAD/PyObjectBase.h>
+#include <Mod/FreeCAD/BoundBox.h>
 #include <string>
 
 
-namespace Base
+namespace FreeCAD
 {
 
 //===========================================================================
@@ -18,7 +18,7 @@ 
 /** The python export class for BoundBox
  */
-class BaseExport BoundBoxPy : public Base::PyObjectBase
+class FreeCADExport BoundBoxPy : public FreeCAD::PyObjectBase
 {
 protected:
     ~BoundBoxPy() override;
@@ -209,8 +209,8 @@ 
 };
 
-}  //namespace Base
-
-#endif  // BASE_BOUNDBOXPY_H
-
-
+}  //namespace FreeCAD
+
+#endif  // FREECAD_BOUNDBOXPY_H
+
+

CMake Error at cMake/FreeCadMacros.cmake:159 (execute_process):
  execute_process failed command indexes:

    1: "Child return code: 1"

Call Stack (most recent call first):
  src/Base/CMakeLists.txt:86 (generate_from_py)

-- Configuring incomplete, errors occurred!

DaemonDave, 2025-03-04:
Same error after 

git clone 

and 

 git submodule update --init --recursive

```

+namespace FreeCAD
 {
 
 //===========================================================================
@@ -18,7 +18,7 @@ 
 /** The python export class for BoundBox
  */
-class BaseExport BoundBoxPy : public Base::PyObjectBase
+class FreeCADExport BoundBoxPy : public FreeCAD::PyObjectBase
 {
 protected:
     ~BoundBoxPy() override;
@@ -209,8 +209,8 @@ 
 };
 
-}  //namespace Base
-
-#endif  // BASE_BOUNDBOXPY_H
-
-
+}  //namespace FreeCAD
+
+#endif  // FREECAD_BOUNDBOXPY_H
+
+

CMake Error at cMake/FreeCadMacros.cmake:159 (execute_process):
  execute_process failed command indexes:

    1: "Child return code: 1"

Call Stack (most recent call first):
  src/Base/CMakeLists.txt:86 (generate_from_py)

```

DaemonDave, 2025-03-04:
Here's the rest of the cmake run I left out because it is failing on the generate.py not on other causes..

[dave@fedora FreeCAD]$ cmake -G "Unix Makefiles" 
CMake Warning:
  No source or binary directory provided.  Both will be assumed to be the
  same as the current working directory, but note that this warning will
  become a fatal error in future CMake releases.

-- Compiler: GNU, version: 12.2.1
-- prefix: /usr/local
-- bindir: bin
-- datadir: share
-- docdir: share/doc/FreeCAD
-- includedir: include
-- libdir: lib64
-- cmake: 3.26.3
-- Compiling with Qt 5
-- find_package() was used to locate fmt version 8.1.1
-- PyCXX found:
--   Headers:  /home/dave/src/FreeCAD/src/3rdParty/PyCXX
--   Sources:  /home/dave/src/FreeCAD/src/3rdParty/PyCXX/CXX
--   Version:  7.1.9
-- -- Found OpenCASCADE version: 7.6.3
-- -- OpenCASCADE include directory: /usr/include/opencascade
-- -- OpenCASCADE shared libraries directory: /usr/lib64
-- Found OpenGL: /usr/lib64/libOpenGL.so   
-- VTK components: CommonCore;CommonDataModel;FiltersVerdict;IOXML;FiltersCore;FiltersGeneral;IOLegacy;FiltersExtraction;FiltersSources;FiltersGeometry;hdf5;RenderingCore;InteractionStyle;RenderingFreeType;RenderingOpenGL2
-- Found OpenGL: /usr/lib64/libOpenGL.so  found components: OpenGL GLX 
-- Found HDF5: /usr/lib64/libhdf5.so (found version "1.12.1") found components: C HL 
-- Found Eigen3: /usr/include/eigen3 (found version "3.4.0") 
-- Check for medfile (libmed and libmedc) ...
-- We guess that libmed was built using hdf5-serial version
-- Set up to compile with Qt 5.15.8
-- Checking Pivy version by importing it in a Python program...
-- Found Pivy 0.6.8
-- Checking Pivy Coin3D version by importing it in a Python program...
-- Found Pivy Coin3D SIM Coin 4.0.0
-- Could NOT find Spnav (missing: SPNAV_LIBRARY SPNAV_INCLUDE_DIR) 
-- Found pip-installed Shiboken2 in /usr/lib64/python3.10/site-packages/shiboken2
-- Found pip-installed PySide2 in /usr/lib64/python3.10/site-packages/PySide2
CMake Warning at cMake/FreeCAD_Helpers/SetupShibokenAndPyside.cmake:128 (message):
  Shiboken2 include files not found, FREECAD_USE_SHIBOKEN automatically set
  to OFF
Call Stack (most recent call first):
  CMakeLists.txt:111 (SetupShibokenAndPyside)

CMake Warning at cMake/FreeCAD_Helpers/SetupShibokenAndPyside.cmake:154 (message):
  PySide2 include files not found, FREECAD_USE_PYSIDE automatically set to
  OFF
Call Stack (most recent call first):
  CMakeLists.txt:111 (SetupShibokenAndPyside)

-- PySide 5.15.7 Python module found at /usr/lib64/python3.10/site-packages/PySide2.

-- Found Matplotlib: /usr/lib64/python3.10/site-packages/matplotlib (found version "3.5.3")
-- Platform is 64-bit, set -D_OCC64
-- Performing Test _flag_found

DaemonDave, 2025-03-04:
CMake Error at cMake/FreeCadMacros.cmake:159 (execute_process):
  execute_process failed command indexes:

    1: "Child return code: 1"

Call Stack (most recent call first):
  src/Base/CMakeLists.txt:86 (generate_from_py)

-- Configuring incomplete, errors occurred!
[dave@fedora FreeCAD]$ history | grep clone
 1080  git clone https://github.com/FreeCAD/FreeCAD.git
 1126  history | grep clone
[dave@fedora FreeCAD]$ !1080
git clone https://github.com/FreeCAD/FreeCAD.git
Cloning into 'FreeCAD'...
remote: Enumerating objects: 495192, done.
remote: Counting objects: 100% (202/202), done.
remote: Compressing objects: 100% (101/101), done.
remote: Total 495192 (delta 131), reused 102 (delta 101), pack-reused 494990 (from 3)
Receiving objects: 100% (495192/495192), 2.04 GiB | 3.01 MiB/s, done.
Resolving deltas: 100% (425627/425627), done.
Updating files: 100% (12211/12211), done.
[dave@fedora FreeCAD]$ git submodule update --init --recursive
[dave@fedora FreeCAD]$ cmake -G "Unix Makefiles" 
CMake Warning:
  No source or binary directory provided.  Both will be assumed to be the
  same as the current working directory, but note that this warning will
  become a fatal error in future CMake releases.

-- Compiler: GNU, version: 12.2.1
-- prefix: /usr/local
-- bindir: bin
-- datadir: share
-- docdir: share/doc/FreeCAD
-- includedir: include
-- libdir: lib64
-- cmake: 3.26.3
-- Compiling with Qt 5
-- find_package() was used to locate fmt version 8.1.1
-- PyCXX found:
--   Headers:  /home/dave/src/FreeCAD/src/3rdParty/PyCXX
--   Sources:  /home/dave/src/FreeCAD/src/3rdParty/PyCXX/CXX
--   Version:  7.1.9
-- -- Found OpenCASCADE version: 7.6.3
-- -- OpenCASCADE include directory: /usr/include/opencascade
-- -- OpenCASCADE shared libraries directory: /usr/lib64
-- Found OpenGL: /usr/lib64/libOpenGL.so   
-- VTK components: CommonCore;CommonDataModel;FiltersVerdict;IOXML;FiltersCore;FiltersGeneral;IOLegacy;FiltersExtraction;FiltersSources;FiltersGeometry;hdf5;RenderingCore;InteractionStyle;RenderingFreeType;RenderingOpenGL2
-- Found OpenGL: /usr/lib64/libOpenGL.so  found components: OpenGL GLX 
-- Found HDF5: /usr/lib64/libhdf5.so (found version "1.12.1") found components: C HL 
-- Found Eigen3: /usr/include/eigen3 (found version "3.4.0") 
-- Check for medfile (libmed and libmedc) ...
-- We guess that libmed was built using hdf5-serial version
-- Set up to compile with Qt 5.15.8
-- Checking Pivy version by importing it in a Python program...
-- Found Pivy 0.6.8
-- Checking Pivy Coin3D version by importing it in a Python program...
-- Found Pivy Coin3D SIM Coin 4.0.0
-- Could NOT find Spnav (missing: SPNAV_LIBRARY SPNAV_INCLUDE_DIR) 
-- Found pip-installed Shiboken2 in /usr/lib64/python3.10/site-packages/shiboken2
-- Found pip-installed PySide2 in /usr/lib64/python3.10/site-packages/PySide2
CMake Warning at cMake/FreeCAD_Helpers/SetupShibokenAndPyside.cmake:128 (message):
  Shiboken2 include files not found, FREECAD_USE_SHIBOKEN automatically set
  to OFF
Call Stack (most recent call first):
  CMakeLists.txt:111 (SetupShibokenAndPyside)

CMake Warning at cMake/FreeCAD_Helpers/SetupShibokenAndPyside.cmake:154 (message):
  PySide2 include files not found, FREECAD_USE_PYSIDE automatically set to
  OFF
[... 62 lines cut ...]
     ~BoundBoxPy() override;
@@ -209,8 +209,8 @@ 
 };
 
-}  //namespace Base
-
-#endif  // BASE_BOUNDBOXPY_H
-
-
+}  //namespace FreeCAD
+
+#endif  // FREECAD_BOUNDBOXPY_H
+
+

CMake Error at cMake/FreeCadMacros.cmake:159 (execute_process):
  execute_process failed command indexes:

    1: "Child return code: 1"

Call Stack (most recent call first):
  src/Base/CMakeLists.txt:86 (generate_from_py)

-- Configuring incomplete, errors occurred!
[dave@fedora FreeCAD]$ git clone https://github.com/FreeCAD/FreeCAD.git

DaemonDave, 2025-03-04:
Full run, nothing left out, still errors on generate.py generating for BoundBoxPy

DaemonDave, 2025-03-04:
Thank you for your help: 

I don't have the latest from a git clone run seconds before?

If I patch with generate.py taken from bleeding edge that might fix the problem?

DaemonDave, 2025-03-05:
Problem solved... then compile started and ...
```

[ 13%] Automatic MOC and UIC for target FreeCADGui
[ 13%] Built target FreeCADGui_autogen
[ 13%] Building CXX object src/Gui/CMakeFiles/FreeCADGui.dir/Inventor/SoFCPlacementIndicatorKit.cpp.o
/home/dave/src/FreeCAD/src/Gui/Inventor/SoFCPlacementIndicatorKit.cpp: In lambda function:
/home/dave/src/FreeCAD/src/Gui/Inventor/SoFCPlacementIndicatorKit.cpp:156:14: error: ‘Color’ is not a member of ‘App’; did you mean ‘Base::Color’?
  156 |         App::Color axisColor(packedColor);
      |              ^~~~~
In file included from /home/dave/src/FreeCAD/src/App/Material.h:27,
                 from /home/dave/src/FreeCAD/src/Gui/Utilities.h:28,
                 from /home/dave/src/FreeCAD/src/Gui/Inventor/SoFCPlacementIndicatorKit.cpp:50:
/home/dave/src/FreeCAD/src/Base/Color.h:89:18: note: ‘Base::Color’ declared here
   89 | class BaseExport Color
      |                  ^~~~~
/home/dave/src/FreeCAD/src/Gui/Inventor/SoFCPlacementIndicatorKit.cpp:172:64: error: ‘axisColor’ was not declared in this scope
  172 |             pcBaseColor->rgb.setValue(Base::convertTo<SbColor>(axisColor));
      |                                                                ^~~~~~~~~
/home/dave/src/FreeCAD/src/Gui/Inventor/SoFCPlacementIndicatorKit.cpp:154:42: warning: unused parameter ‘packedColor’ [-Wunused-parameter]
  154 |                                 uint32_t packedColor,
      |                                 ~~~~~~~~~^~~~~~~~~~~
/home/dave/src/FreeCAD/src/Gui/Inventor/SoFCPlacementIndicatorKit.cpp: In member function ‘SoSeparator* Gui::SoFCPlacementIndicatorKit::createPlaneIndicator()’:
/home/dave/src/FreeCAD/src/Gui/Inventor/SoFCPlacementIndicatorKit.cpp:269:28: error: ‘Color’ is not a member of ‘App’; did you mean ‘Base::Color’?
  269 |     const auto gray = App::Color(0.75f, 0.75f, 0.75f);
      |                            ^~~~~
/home/dave/src/FreeCAD/src/Base/Color.h:89:18: note: ‘Base::Color’ declared here
   89 | class BaseExport Color
      |                  ^~~~~
gmake[2]: *** [src/Gui/CMakeFiles/FreeCADGui.dir/build.make:3174: src/Gui/CMakeFiles/FreeCADGui.dir/Inventor/SoFCPlacementIndicatorKit.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:8628: src/Gui/CMakeFiles/FreeCADGui.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
[dave@fedora FreeCAD]$ 

```

furgo16, 2025-03-05:
Please post the output of `git log -1` as requested on https://github.com/FreeCAD/FreeCAD/issues/19956#issuecomment-2699014403. As blind guesses, you might not have the lastest git revision or you've not started a clean build.

DaemonDave, 2025-03-05:
dave@node-1w7jr9uohrzglgatr4j05vzhi bindings]$ git log -1
commit 7f1c43a335294cc7c8af7962eb9367550c2751af (HEAD -> main, origin/main, origin/HEAD)
Merge: a9b7490a82 1ce7c678bb
Author: Chris Hennes <chennes@pioneerlibrarysystem.org>
Date:   Sun Mar 2 16:36:40 2025 -0600

    Merge pull request #19636 from kadet1090/color-in-base
    
    Base: Move App::Color to Base
[dave@node-1w7jr9uohrzglgatr4j05vzhi bindings]$

furgo16, 2025-03-05:
Thanks. Your git clone is at commit https://github.com/FreeCAD/FreeCAD/commit/7f1c43a335294cc7c8af7962eb9367550c2751af, which is quite behind. I think that commit was known to break the build, although I'm not 100% certain.

In any case, please use the `git pull` command to update your clone to the last upstream revision, and double-check you are indeed at the latest revision. Then you can build and report back.

DaemonDave, 2025-03-05:
In other words: How do I turn on the CXXFLAGS CPPFLAGS CFLAGS flag for 

 -fPIC

 into the cmake configuration?

furgo16, 2025-03-06:
Thanks for the formatted log. I think at this point this issue can be safely closed (original error no longer appearing), and I'd recommend to continue the conversation as a support request on the forum, where more folks experienced with the build process and Fedora can help debugging and diagnosing the build failure: https://forum.freecad.org/viewforum.php?f=4&sid=37a3803b21a0cd834ec0ecfb3c0bb9fc