FreeCAD/FreeCAD #19957: Pre-commit.ci auto fixes are making changes that break the code

LarryWoestman, 2025-03-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In PR #19955, the pre-commit.ci auto fixes are breaking the submitted code.

I noticed it removing spaces from the end of the line inside of a multi-line string.  This breaks the test cases which are based on the multi-line strings matching the output of the tests.

Is there a way to tell the pre-commit.ci code not to make those changes in this case?

### Full version info

```shell
The build occurred on 20250302 but the possibility of this issue has probably been around for a while.

I suspect that this issue would affect any part of the check-in process where the pre-commit.ci tool is used on a file that happens to contain multi-line strings.
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-03-04:
Have you tried adding that specific Python file to the `exclude` section of https://github.com/FreeCAD/FreeCAD/blob/main/.pre-commit-config.yaml ? To the best of knowledge you can't exclude specific lines it has to be file(s) or folder(s).

LarryWoestman, 2025-03-04:
That particular change appears to be equivalent and is probably caused by the auto fixes using a different maximum line length than I am.  The last time I looked at the config files for the auto-fix it was using 100 characters for the maximum line length, so that is what I have configured the black formatter to do.  However, I continue to see auto-fix changes (related to line length like this one) that I can't explain.

The problem is on lines 388 (384 in the revised file) and 408 (404 in the revised file).  The file currently has a space at the end of a those lines, which are in the middle of multi-line python strings.

I could modify my code (and tests) to work around these particular cases, but the issue would still remain to bite someone else.

It appears to me that the current auto-fix code doesn't handle a space at the end of a line in a python multi-line string appropriately.  Unlike in many cases, such a space "matters" because it is in the middle of a string, and it should not be removed.

LarryWoestman, 2025-03-04:
I haven't tried to make this specific file a "special case", yet.   I can modify my code (and tests) to work around this issue, but that still leaves the issue present ready to cause problems later.

Thanks for the suggestion.

LarryWoestman, 2025-03-04:
I am testing the output of postprocessing g-code and comparing what "should be there" with what the postprocessor actually generates.

Yes, I could modify the postprocessor code to not generate a space at the end of a line, but that doesn't help fix the issue with the auto-fix code.  It just leaves a problem waiting for the next similar case in the future, which might not be so easy to notice when it happens.

Thanks for the suggestion on using ```#fmt:  off/on```.  That might be a work-around for my particular case, but still doesn't address the auto-fix issue.

LarryWoestman, 2025-03-04:
It appears to me that the formatter doesn't recognize that it is in the middle of a multi-line python string.  That is a problem waiting to happen in the future that needs to be fixed (assuming that I understand the problem correctly).

LarryWoestman, 2025-03-04:
I have lots of options for workarounds :-)  I will use one of them to let me get on with my work, but the issue will still remain and still need to be fixed, or it will continue to need workarounds in the future.

FEA-eng, 2026-08-27:
@LarryWoestman Please confirm if it's still relevant

luzpaz, 2026-09-08:
What's the verdict here ?