FreeCAD/FreeCAD #19958: Sketcher: B-Spline to B-Spline tangency at the ends not working

Shkolik, 2025-03-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

B-Spline to B-Spline tangency at the ends not working. Solver complaints about "tangent" constraint as redundant. It has workaround with construction line trough 2 control points and coincidence point, but it doesn't help when one of the splines is external geometry.

[image]

Test file:

[spline to spline tangency.FCStd.zip](https://github.com/user-attachments/files/19046222/spline.to.spline.tangency.FCStd.zip)

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40443 (Git)
Build type: Release
Branch: all_fixes
Hash: 05aa0558e9a21b28ac5dd1757c607c7ae8c3272c
Python 3.12.7, Qt 6.8.1, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical/physical DPI: 96/113.414
Installed mods: 
  * AirPlaneDesign 0.4.1
  * CurvedShapes 1.0.13
  * Curves 0.6.58
  * fasteners 0.5.37
  * Foamcut 0.1.6
  * freecad.gears 1.3.0
  * Marz 0.1.15
  * OpenTheme 2024.9.1
  * PieMenu 1.11.2
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-03:
How did you apply the constraint? Did you select both points or splines?

Shkolik, 2025-03-03:
> How did you apply the constraint? Did you select both points or splines?

Both splines and common point. 
More than that - I can not make straight line tangent to the splene at end point. It works fine anywhere on spline except endpoints.

maxwxyz, 2025-03-03:
@Shkolik as you can see it works as expected:

https://github.com/user-attachments/assets/27d521ac-f0ec-4b14-84ac-f66e0021116c

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40444 (Git) Conda
Build type: Release
Branch: main
Hash: 089605bb4db98668661bf56db3f5f5bb4f7f0d87
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/108.888
Installed mods: 
  * Cables 0.1.3
  * CfdOF 1.29.8
  * CurvedShapes 1.0.13
  * Curves 0.6.58
  * dodo 1.0.1
  * fasteners 0.5.37
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.1.2
  * parts_library
  * Rocket 4.1.1
  * sheetmetal 0.7.20
  * Silk 0.1.5

```

Shkolik, 2025-03-03:
Ok, so workflow wasn't clear then. For tangency to b-spline in point other than endpoints we have to select both curves and common point, but for ends tangency - only endpoints, and it's not stated anywhere. Also if splines already connected with coincidence constraint it doesn't seem to replace it with tangent as solver does for other types of curves. I'll test this cases in a couple hours and will share my findings, but at least we have a problem with not clear/not consistent workflow.

Shkolik, 2025-03-03:
Ok, I checked possible combinations and looks like code is there and it's doing it's job. The only complaint about not consistent workflow.
For example if we want to make line tangent to ark we can:
1) when arc not connected to the line - select both and apply tangent constraint. Line will be tangent to the arc (to the underlining circle, to be precise), but they will remain not connected.
2) connect arc endpoint to line endpoint by coincidence constraint, then select both and apple tangent constraint. Coincident constraint will be removed and replaced with tangent constraint. Curves will remain connected. 

With B-splines we have to different workflows and it's kinda misleading:
1) if we want to make curve tangent to b-spline and their ends connected we have to select endpoints and apply tangent constraint - just like on video. But if curves already connected at endpoints - error "Wrong selection. Wrong number of selected objects!" appear. And error not saying how to fix it.
2) if we want make curve tangent to b-spline in any other that endpoint point, we have to select both curves and point where they will be tangent - that's fine, but totally different workflow.

What I suggesting:
1) Be able to make curve to b-spline tangent when selecting 2 connected curves and apply constraint - solver should try to replace coincidence with tangency as it works for all other types of curve.
2) If curves not connected there are 2 ways. We can behave just like now, but instead of selecting endpoints specifically let user to select curves and make them connected at closest endpoints and tangent, or make curves tangent, but do not move endpoints and not connect them - just like other types does.

And for sure error description should be more descriptive.  Instead of "Wrong selection. Wrong number of selected objects!" should be "Wrong selection. To make selected objects tangent do [instruction how properly select and what exactly select]".

FEA-eng, 2026-04-09:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.2.0dev.20260407 (Git shallow)
Build date: 2026/04/07 22:56:13
Build type: Release
Branch: (HEAD detached at 6db886fe0)
Hash: 6db886fe0e79bf060ffeaa50198da25702a17e31
Python 3.11.15, Qt 6.8.3, Coin 4.0.8, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

FEA-eng, 2026-10-25:
Still relevant

```
OS: Windows 11 build 26300
Architecture: x86_64
Version: 27.1.0dev.20261020 (Git shallow)
Build date: 2026/10/20 22:15:52
Build type: Release
Branch: (HEAD detached at 9946038aa)
Hash: 9946038aace97af288eb28f2727a169f9d1158c7
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Gesture/Trackball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 621.01/NVIDIA Corporation/NVIDIA GeForce RTX 5080/PCIe/SSE2
```