FreeCAD/FreeCAD #19963: Sweep and Frenet AdditivePipe on arc Section create broken geometry

ElementW, 2025-03-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I used a [image] *Part Section* of two [image] *Part Extrude*d sketches featuring arcs to create a 3D path to extrude a profile along.
Both [image] *Part Sweep* and [image] *PartDesign AdditivePipe* in Frenet mode (and some other modes too) create a completely broken extrusion along the path, with excessive degenerate geometry that bloats the FCStd file.

#17762 and #5651 could be related but they're about Bézier and B-splines respectively, not arcs, not sure how FC/OCCT handles those as primitives.

Left/orange is the sweep, right/green is the additive pipe.
In addition computing the geometry for both takes a long time, but the pipe is especially long for almost the same output geometry.

[image]

[broken.FCStd.zip](https://github.com/user-attachments/files/19052373/broken.FCStd.zip)

### Full version info

```shell
OS: Void Linux (KDE/xcb)
Architecture: x86_64
Version: 1.1.0dev.40444 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 089605bb4db98668661bf56db3f5f5bb4f7f0d87
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/92.6073
Installed mods: 
  * fasteners 0.5.31
  * OpenTheme 2024.9.1
  * FreeCAD-Ribbon 1.4.3 (Disabled)
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-09:
@pieterhijma FYI (I hope it's OK that I ping you on such issues)

pieterhijma, 2025-03-14:
> [@pieterhijma](https://github.com/pieterhijma) FYI (I hope it's OK that I ping you on such issues)

Thanks, sure!

tomate44, 2025-05-18:
The bug is caused by the use of a Section object as sweep path.
More precisely, the bug is located on the two degree 1 BSpline curves of the section.
This is not a problem with arcs.

ElementW, 2025-05-18:
@tomate44 I'm inclined to believe this, however my few attempts at using a Draft 3D b-spline to create a similar shape did not yield broken results like this; so this may be specific to splines created as a result of a section (of arcs or otherwise).

tomate44, 2025-05-18:
I agree. BSplines created with Draft WB are probably cubic, so the Frenet Coordinate System is usually defined all along the curve. On the other hand, the section tool often generates linear BSplines (degree 1) that are not very good sweep paths.

FEA-eng, 2025-10-01:
Still relevant

Also, I appreciate the OP's idea to add icons of the tools mentioned in the description.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/30 23:55:55
Build type: Release
Branch: main
Hash: b08a0f87d68dd38cfcf39046209a7075adf11576
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```