FreeCAD/FreeCAD #19973: Addon Manager: Update progress is not legible but window too wide

maxwxyz, 2025-03-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When updating add-ons the new window is by default way too large, spanning almost over the entire width of my 24" screen.
It would be ok if the content makes use of this but neither the name of the addon nor the progress is legible:

[image]

@chennes FYI

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40444 (Git) Conda
Build type: Release
Branch: main
Hash: 089605bb4db98668661bf56db3f5f5bb4f7f0d87
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/108.888
Installed mods: 
  * Cables 0.1.3
  * CfdOF 1.29.8
  * CurvedShapes 1.0.13
  * Curves 0.6.58
  * dodo 1.0.1
  * fasteners 0.5.37
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.1.2
  * parts_library
  * Rocket 4.1.1
  * sheetmetal 0.7.20
  * Silk 0.1.5
```

### Subproject(s) affected?

Addon Manager

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

obelisk79, 2025-03-03:
Unable to duplicate this issue. This may be a windows-specific issue.

[image]

kaiwas, 2025-03-03:
Probably this is to fit all these buttons.
I was very surprised today

[image]

maxwxyz, 2025-03-03:
what buttons, oh wow

kaiwas, 2025-03-03:
> oh wow. I don't have buttons, what do they say?

I will now reboot my computer and try to reproduce the problem (in English)

maxwxyz, 2025-03-03:
Maybe since https://github.com/FreeCAD/FreeCAD/pull/19791 was merged

chennes, 2025-03-03:
Interesting, it's trying to show every single possible button in the button box. I can't reproduce, but based on that I will take a look.

kaiwas, 2025-03-03:
appears during update

https://github.com/user-attachments/assets/4b72f897-459c-4ed3-b48c-2fc5c379e810

chennes, 2025-03-03:
That's so fascinating: the button box is supposed to start out with no buttons in it at all (it's set in the UI file as "NoButton"). I can change the way I do that so instead have the Close button in the box, but hide the whole QButtonBox. I'm betting this is a Qt-version-specific thing, rather than an OS-specific thing.

kaiwas, 2025-03-04:
> a Qt-version-specific thing

This effect was not present on previous builds.
My KDE-plasma version has not changed for quite a long time.

closed 2025-03-04 by chennes

chennes, 2025-03-05:
> > a Qt-version-specific thing
> 
> This effect was not present on previous builds. My KDE-plasma version has not changed for quite a long time.

I don't mean it was a problem with the build: I mean that it's specific to the version of Qt in the build. Qt6 builds did not exhibit the problem because Qt6 changed the scoping of their enumerations. When I edited the UI file with a Qt6 version of designer, it used the Qt6 naming, which Qt5 doesn't recognize. The PR in question simply switched back to the Qt5 naming, which works in both.

PR #19975 (2025-03-04): Addon Manager: Adjust use of QDialogButtonBox
@maxwxyz can you see if this fixes #19973?