FreeCAD/FreeCAD #19980: Crash while taking a measurement

m1cha1, 2025-03-04:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I have a crash using Measure tool
1. Open PartDesignExample
2. Click on Measure tool
3. Click on random wall - it will display area
4. Click on different wall-crash

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/141.585
Installed mods: 
  * Cables 0.1.3
  * Curves 0.6.58
  * Design456 0.0.1
  * fasteners 0.5.37
  * FreeCAD-Telemetry 1.0.0
  * freecad.gears 1.3.0
  * sheetmetal 0.7.21
  * workfeature
  * workfeature-macro
```

### Subproject(s) affected?

Measurement

### Anything else?

Tested in Safe Mode
I have the Telemetry add-on.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-04:
Please provide a .zip example file

maxwxyz, 2025-03-04:
Does it also crash on the stable 1.0 version for you?

maxwxyz, 2025-03-04:
OK I can confirm this issue.

@kpemartin might this be related to https://github.com/FreeCAD/FreeCAD/issues/19881 ?
@hlorus FYI

kpemartin, 2025-03-04:
I have a fix for #19881 almost ready, just doing a last compile and test.
I'll try this with the master branch later this morning when I have a chance to give it a fresh build. Seems unlikely to be related, but I've thought that before... I see (using the stable 1.0 release) that the label for the measurement object is the actual measurement (e.g. total area) so it is perhaps possible that the label transiently has too many trailing digits, causing the problem #19881 fixes. OTOH the measure tool does not seem to actually create a DocumentObject until you Save the measurement, and also in this example the measurement is an area so it always ends with "^2" which is only a single digit.

kpemartin, 2025-03-04:
I believe this to be entirely unrelated to #19881.
It still occurs in my branch that contains the fix for that issue, and the error is a stack overflow in `painter.end()` called from line 488 of src/Gui/SoTextLabel.cpp.

That being said, the infinite recursion seems to involve code trying to set the Label to something like `Area: 1.60 cm^001` and the reaction that takes place in the handling of the onChanged event handler in `MeasureArea::onChanged`

That odd label text implies that the value one would expect to set `Area: 1.60 cm^2` was found as an existing label and has been uniquefied. It would appear that the reaction to the label change wraps back to `ViewProviderMeasureBase::updateData` who stubbornly tries to set the label to `Area: 1.60 cm^2` again. This would mean the problem was introduced with #18676

maxwxyz, 2025-03-04:
Could this be related to https://github.com/FreeCAD/FreeCAD/issues/19709 then?

captain0xff, 2025-03-04:
I was also looking at this and the code seems to be segfaulting at https://github.com/FreeCAD/FreeCAD/blob/fc35a58346aee9941424150bf25e7657218ef517/src/Gui/Tools.h#L48. I tried to understand your reasoning above but am not sure why QFontMetrics::horizontalAdvance (called from https://github.com/FreeCAD/FreeCAD/blob/fc35a58346aee9941424150bf25e7657218ef517/src/Gui/SoTextLabel.cpp#L433) will segfault unless something is wrong with the font. I am interested in working on this issue and can you kindly guide me where should I look to solve this.
I am just a student trying to start contributing to freecad so sorry if I am speaking some nonsense :)

kpemartin, 2025-03-04:
I can reproduce this problem (with a little more difficulty) in the 1.0.0 release.

In general the stack infinite recursion occurs if the label set on the measurement object (which is a text form of the measurement) is found to already exist in the document. This generally requires measuring something that has the same measure as what has been previously measured within the same Measure command (without Save or Close). For this particular document, the areas of the tops of those lugs are the same, fulfilling that condition.

In this Issue as reported, the label appears to exist though it shouldn't because the measurement logic does not actually remove the old measurement objects, instead setting a flag on them, I assume so they get deleted later.

In 1.0.0 I tried to reproduce the problem per se but the ghost labels are not visible so re-measuring something with the same measure works. Trying to provoke the bug, I saved a measurement, then restarted the Measure command and measured the same object, and it still worked. This turns out to be because the Label generated by the Measurement is prefixed with the object Name of the measurement, which ensures the label is unique. Well, not really, since the user can manually rename existing measurement objects to force a collision with an upcoming measurement.

To reproduce in 1.0.0:
Load the sample PartDesignExample document.
Start the Measure tool
Select a surface (for instance the top of one of the lugs at the top of the part)
Save the measurement
Close the Measure task pane
Scroll down in the Model tree view, and open the Measurements node
You will see one measurement with the label "Area: 1.60 cm^2" (assuming the top of the lug was selected)
In the Data tab modify this label to be "Area001: 1.60 cm^2"
Start the Measure tool
Select the same surface (or another having the same area)
Boom

So it is an existing problem in 1.0.0, which has been given a new manifestation by #18676.

kpemartin, 2025-03-04:
The measurement code may have stuck the object Name into the Label a long time ago as an ultimately (as now observed) attempt to avoid the infinite recursion.
When I saw the text "Area" in the label I just thought it was pointing out that the measurement is an area, but it was really just putting in the Name, which happened to be "Area"

Perhaps a better way to do this would be for the measurement object to override the `allowDuplicateLabels()` property to return true, thus allowing the measurement to have the same label as other objects, and rather than stuffing the Name into the Label, just put in the model name used for the Name, which in this case would always be "Area", but could also be "Radius", "Center_of_Mass", etc.

This should fix both manifestations of the infinite recursion, though it still leaves the fact that the new Unique-name management will see ghost labels that the old unique-name management did not, which is a separate thing for me to look into.

kpemartin, 2025-03-04:
Oh, a third thing to look into is why having the duplicate-name code change the proposed label causes infinite recursion.

maxwxyz, 2025-03-05:
Not sure if relevant but the measurement label updates its name when the value it measures changes

kpemartin, 2025-03-05:
@maxwxyz I'm pretty sure that object Names are invariant once the object is part of a document. This is stored in `DocumentObject::pcNameInDocument` which points to the `std::string` name that is the key in the document's objectMap, and is only ever set to null when the object is removed from the document, which is not happening in this case. On the other hand, the object's Label is changed to reflect the actual measurement text.

kpemartin, 2025-03-05:
I've found why #18676 exacerbates the problem (that is, causes it without the gymnastics required to demonstrate it in 1.0.0): In `Document::removeObject` (and also `_removeObject`, perhaps), the object being removed is passed to `activeUndoTransaction->addObjectNew`. In this case the tx already contains a Del entry for the same object placed there by newObject the last time the selection to be measured was changed, so `addObjectNew` just unceremoniously deletes the document object. Essentially the overall "Do" created then deleted an object, so the "Undo" can skip undoing both operations.
After this call `removeObject` tries to unregister the object's Label, which is of course empty now (and, in fact, in freed memory).
So essentially any time an object is created and then deleted under the same Undo transaction, its Label will be left in the list of registered Labels causing phantom Label collisions in the future.
`removeObject` should unregister the label before the object might be deleted by `activeUndoTransaction->addObjectNew` (it does not happen if `d->rollback`).

Following that, `removeObject` does a linear search of the Document's `objectArray` to remove any reference to the object. This is comparing a pointer to freed memory for equality but not dereferencing it.

closed 2025-03-06 by chennes

kpemartin, 2025-03-06:
Should this be fixed in 1.0 as well? Should that be a separate Issue?

kpemartin, 2025-03-06:
Submitted #20033 to correct this in the releases/FreeCAD-1-0 branch

PR #20016 (2025-03-06): Fix the Measure Tool which will crash the app with a stack overflow if you measure the same amount twice
Fixes #19980

PR #18676 changed the management of Label uniqueness. A bug in that change meant that if a DrawingObject were created, then deleted, in the same transaction, the record of its Label would remain, and subsequent attempts to use the same Label might find the intended Label changed to make it unique.

The Measure tool is quite insistent that the Label of its object be of the form "some text: actual measurement". If it attempts to set this but uniqueness forces the text to be "som [... cut]