FreeCAD/FreeCAD #19983: Polar Pattern- entering the Occurrences value via keyboard too quickly results in NO change

JDMils, 2025-03-04:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I posted my findings on the forum here:
https://forum.freecad.org/viewtopic.php?p=814622#p814622

In summary, I have a polar pattern of a triangle cut out of a ring. If I open the polar pattern end edit the Occurrences value to any value then press ENTER quickly, the number of occurrences does NOT change and visually the number of occurrences on the model have not changed yet the Occurrences value shows the new value entered via the keyboard.

To reproduce this issue:
* Double-click the Polar Pattern object to open the PP dialog.
* In the Occurrences textbox, enter say 100 and press ENTER quickly. The model does not change in appearance as it still flouts the original PP Occurrences value.
* Click the PP in the tree and check the Occurrences value- it shows the new value entered via the keyboard.

I can work around this issue by:
* Entering the Occurrences value via the keyboard and wait for the model to change and reflect the new value.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40317 (Git) Conda
Build type: Release
Branch: main
Hash: b67ab979afdba808fc03196faf9c5f8830df471f
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/Australia (en_AU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Logical/physical DPI: 96/106.57
Installed mods: 
  * A2plus 0.4.68
  * Assembly4 0.50.15
  * Curves 0.6.58
  * fasteners 0.5.36
  * Gridfinity 0.8.0
  * lattice2 1.0.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-03-22:
See forum thread for discussion https://forum.freecad.org/viewtopic.php?p=814743#p814743

FEA-eng, 2025-09-30:
Still relevant

[image]

@PaddleStroke FYI

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

PaddleStroke, 2025-09-30:
I can reproduce. Pressing Enter validates the task. But the feature is not recomputed. As tritao mentioned the UI triggers a timer when you change the number of occurence and recompute the feature only after timer is finished. To avoid recomputing on every change of the value (say you use the scroll of the mouse for example).
What happens here is that pressing enter validates the task which prevent the timer from triggering.