FreeCAD/FreeCAD #19987: Simple FEM thermomechanical model produces massive memory leak --> crash

RiccaDS, 2025-03-04:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In this simple thermomechanical model I attach, when I write the inp file and run calculix, memory usage increases indefinitely, eats up all swap memory and then crashes system. I have no clue. 

[test.zip](https://github.com/user-attachments/files/19070739/test.zip)

### Full version info

```shell
OS: Ubuntu 24.10 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/140.99
Installed mods: 
  * Curves 0.6.55
  * Manipulator 1.5.7
  * Estimate 0.1.2
  * fasteners 0.5.37
  * ThreadProfile 1.92.0
  * Defeaturing 1.2.3
  * A2plus 0.4.68
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

FEA-eng, 2025-03-04:
> when I write the inp file and run calculix

Since it writes the .inp file normally and the issue occurs after starting CalculiX, it's a problem on the solver's side. Running this input file outside of FreeCAD (using ccx binary) reveals a lot of overconstraint errors. It's because of redundant tie constraint. You have a compound with no gap so the mesh is continuous and tie constraint is not necessary. Disabling it fixes the problem. See the first note here: https://wiki.freecad.org/FEM_ConstraintTie#Notes

RiccaDS, 2025-03-04:
> Since it writes the .inp file normally and the issue occurs after starting CalculiX, it's a problem on the solver's side. Running this input file outside of FreeCAD (using ccx binary) reveals a lot [...]

Yes, correct, I forgot about the gap/nogap behavior. Sorry I posted here but the forum was unavailable. Thank you for helping.

closed 2025-03-04 by maxwxyz